solo-io/packer-plugin-arm-image #1: packer build command needs sudo.

bunchc, 2018-01-09:
**Command:**

```packer build example.json```

**Expected:**

Completed packer build

**Result:**
```
==> arm-image: Error executing command 'kpartx -s -a output-arm-image/image': exit status 1
==> arm-image: Stderr: /dev/mapper/control: open failed: Permission denied
==> arm-image: Failure to communicate with kernel device-mapper driver.
==> arm-image: Incompatible libdevmapper (unknown version) and kernel driver (unknown version).
==> arm-image: device mapper prerequisites not met
==> arm-image:
==> arm-image: Error executing command 'kpartx -d output-arm-image/image': exit status 1
==> arm-image: Stderr: /dev/mapper/control: open failed: Permission denied
==> arm-image: Failure to communicate with kernel device-mapper driver.
==> arm-image: Incompatible libdevmapper (unknown version) and kernel driver (unknown version).
==> arm-image: device mapper prerequisites not met
==> arm-image:
Build 'arm-image' errored: Error executing command 'kpartx -d output-arm-image/image': exit status 1
Stderr: /dev/mapper/control: open failed: Permission denied
Failure to communicate with kernel device-mapper driver.
Incompatible libdevmapper (unknown version) and kernel driver (unknown version).
device mapper prerequisites not met

==> Some builds didn't complete successfully and had errors:
--> arm-image: Error executing command 'kpartx -d output-arm-image/image': exit status 1
Stderr: /dev/mapper/control: open failed: Permission denied
Failure to communicate with kernel device-mapper driver.
Incompatible libdevmapper (unknown version) and kernel driver (unknown version).
device mapper prerequisites not met

==> Builds finished but no artifacts were created.
```

**Work around:**

sudo packer build example.json

bunchc, 2018-01-09:
I... I am an idiot. It is right there in the readme. Apologies.

closed 2018-01-09 by bunchc

yuval-k, 2018-01-09:
First issue is always a cause for celebration!
@bunchc any feedback is welcome!

bunchc, 2018-01-09:
So far so good. It took some doing to get it going, but was able to make my first image just fine. I'm testing a vagrant file / bash combo to make setting up the environment a bit easier.

yuval-k, 2018-01-09:
Nice! Did I mention we love PRs ;-) ?