solo-io/packer-plugin-arm-image #7: Vagrantfile won't mount /vagrant synced folder (and enabling it will fail in Virtualbox)

tommie-lie, 2018-03-16:
Unfortunately, merging my PR #4 introduced a grave regression: The generic images have the default synced folder disabled. It is easy to enable it again by adding
```
  config.vm.synced_folder "./", "/vagrant", disabled: false
```
to the Vagrantfile, but this causes issues if the VirtualBox provider is used:
* The box does not have `virtualbox-guest-utils` installed which, means `mount.vboxsf` is missing and the necessary kernel modules to mount VirtualBox shared folders are not automatically loaded.
* Manually installing `virtualbox-guest-utils` **before** Vagrant tries to mount the synced folder is not possible. This is a common problem and there are many issues filed for this. All workarounds that have been proposed don't work anymore because Vagrant has evolved so erratically that features that once existed are still visible in some sourcecode (https://github.com/hashicorp/vagrant/blob/master/plugins/providers/virtualbox/synced_folder.rb#L38) but are rendered unusable somewhere else.

There are two options to solve that:
* `vagrant plugin install vagrant-vbguest` which automatically installs the matching version of the VirtualBox guest additions before Vagrant mounts the folders and probably solves a lot of other problems with VirtualBox.
* Use a different base box, e.g. https://app.vagrantup.com/ubuntu/boxes/xenial64 only when using VirtualBox:
```
  config.vm.provider :virtualbox do |vb, override|
    override.vm.box = "ubuntu/xenial64"
    # disable the generation of ubuntu-xenial-16.04-cloudimg-console.log file
    # https://betacloud.io/get-rid-of-ubuntu-xenial-16-04-cloudimg-console-log/
    vb.customize [ "modifyvm", :id, "--uartmode1", "disconnected" ]
  end
```

libvirt (the reason I changed base box in the first place) is not affected by all of this. Could someone with a VMware or Parallels license please test whether they have a similar issue?

Also, I'd like to know if the above makes sense and if using `ubuntu/xenial64` works for more than my version of VirtualBox (5.2.8) to avoid making the same mistake again as last time.

ufuchs, 2018-03-17:
Maybe this is a solution:
https://github.com/boxcutter/ubuntu
Best regards

yuval-k, 2018-03-17:
Unfortunately I don't use Vagrant or virtual box so have no way to test this. To my understanding from discussion above, should this Vargrant file work? : 
```
# -*- mode: ruby -*-
# vi: set ft=ruby :

Vagrant.configure("2") do |config|
  config.vm.box = "generic/ubuntu1604"
  config.vm.provision "shell", :path => "provision.sh", privileged: false
  config.vm.provider :virtualbox do |vb, override|
    override.vm.box = "ubuntu/xenial64"
    # disable the generation of ubuntu-xenial-16.04-cloudimg-console.log file
    # https://betacloud.io/get-rid-of-ubuntu-xenial-16-04-cloudimg-console-log/
    vb.customize [ "modifyvm", :id, "--uartmode1", "disconnected" ]
  end
end
```

tommie-lie, 2018-03-17:
@ufuchs: The boxcutter boxes are great as a basis for when you want to develop your own boxes with packer, but they are not (to my knowledge) available on the Vagrant Cloud, so they are not very useful for the purpose at hand. Also, they don't have libvirt boxes.

@yuval-k: Almost, the synced folder has to be re-enabled. tommie-lie/packer-builder-arm-image@b39cec37 was tested on libvirt and Virtualbox and works for both. I was hesitant to file a PR because I wanted to give it more testing but feel free to merge if you like.

P.S.: Some more Vagrantfile feature enhancements to come early next week ;-)

yuval-k, 2018-03-17:
Sounds good, I'll merge it for now so we will have something working

yuval-k, 2018-03-17:
sounds great re: vagrant enhancements, thank you! 
feel free to suggest any other feedback in issues \ our slack channel

yuval-k, 2018-03-17:
fixed by 2738b03a61f15a0790f3781ceb3df9373139f69d

closed 2018-03-17 by yuval-k