solo-io/packer-plugin-arm-image #12: Does this work with archlinux arm?

bcomnes, 2018-09-15:
Any idea if this would work with the archlinux arm images https://archlinuxarm.org/platforms/armv8/broadcom/raspberry-pi-3 

Apologies, I'm quite new to packer and still trying to understand the dependencies between the images and providers.

yuval-k, 2018-09-17:
It should work in any type of image, though some settings may need to be changed from the default.
The full settings are in here:  https://github.com/solo-io/packer-builder-arm-image/blob/2a33747c943b22b1e6642a6f226f1ab88f81fb42/pkg/builder/builder.go#L35
One thing that comes to mind that `image_mounts` might be different.

Did you have a different experience?

bcomnes, 2018-09-17:
No, just wanted to check.  Thanks for the pointer!  I'll PR any undocumented discoveries back that I find.

bcomnes, 2019-01-07:
Sorry to bug you again, finally getting around to playing with this.

Do you have any pointers on examples of configuring `image_mounts`?  For context, the raspi arch image is a tar.gz with a folder called `/boot` that needs to go into the first partition.  (see https://archlinuxarm.org/platforms/armv8/broadcom/raspberry-pi-3 for more details)

The tar file has the following structure:

```
bin  boot dev  etc  home lib  mnt  opt  proc root run  sbin srv  sys  tmp  usr  var
```

yuval-k, 2019-01-17:
the idea of image_mounts is to say where to mount the paritions in the sdcard, in order.
i.e. if image mounts is `["/boot", "/"]` first partition will be mounted as /boot, and the second one as the root (/)

from these logs it seems that the sd card has zero partitions - can you confirm that this is the case?
if this is the case, what is the expected behavior?

yuval-k, 2019-01-17:
in case you atr trying to debug, the code that parses out partitions is in here:
pkg/builder/step_map_image.go

bcomnes, 2019-01-17:
Thank you for the hints!

Does it expect to see / write to an SD card during the `packer build` phase?  Maybe I just have a conceptual misunderstanding here.  I expected that it downloads the image, modifies it with the provisioners and then creates a new image file from the results that you then write to an SD card separately.  e.g. the image creation was separate from writing / partitioning the SD card.

yuval-k, 2019-01-21:
your idea of how it works is correct. 
the paritions are needed to correctly mount it locally for modifications.
i.e. in order for the regular packer provisioners to work, i need to mount the image to the local file system. and in order to do that, i need to know the mount points of the partitons.

yuval-k, 2019-01-21:
added some text to clarify: https://github.com/solo-io/packer-builder-arm-image#operation

bcomnes, 2019-01-21:
Cool thank you!  Let me take another stab at it.

yuval-k, 2019-01-21:
from the instructions here `https://archlinuxarm.org/platforms/armv8/broadcom/raspberry-pi-3` it seems that the regular rpi patitions are used, i.e. `["/boot", "/"]`

yuval-k, 2019-01-21:
I think I understand your issue. the tar ball they provide is not a disk image. they do provide instructions on how to make it into a disk image.
Since this program was designed with disk images in mind, it will not work for this use case without modifications.

bcomnes, 2019-01-28:
Ah ok,  that is probably the issue.  I'll play around.  Thank you for the additional pointers + constraints!

bcomnes, 2019-01-28:
> they do provide instructions on how to make it into a disk image.

Do you recall where you found those?  I'm not finding their specific recommendations.

bcomnes, 2019-01-29:
https://github.com/bcomnes/raspi-packer/blob/master/build-img.sh 

Basic iso creation. Seems to be working now, last step is to flash and test!

bcomnes, 2019-01-29:
Fails at 

```
58.17s$ sudo ./packer build arch-raspi.json
arm-image output will be in this color.
Warnings for build 'arm-image':
* A checksum type of 'none' was specified. Since ISO files are so big,
a checksum is highly recommended.
==> arm-image: Downloading or copying Image
    arm-image: Downloading or copying: file:///home/travis/gopath/src/github.com/bcomnes/raspi-packer/arch-linux.img
==> arm-image: Copying source image.
    arm-image: Speed:  134.97 MB/s
    arm-image: Progress: 94.28%
    arm-image: mappping output-arm-image/image
==> arm-image: kpartx -s -a -v output-arm-image/image
==> arm-image: partitions: [/dev/mapper/loop0p1 /dev/mapper/loop0p2]
    arm-image: Mounting: /dev/mapper/loop0p2
    arm-image: Mounting: /dev/mapper/loop0p1
==> arm-image: Mounting additional paths within the chroot...
    arm-image: Mounting: /proc
    arm-image: Mounting: /sys
    arm-image: Mounting: /dev
    arm-image: Mounting: /dev/pts
    arm-image: Mounting: /proc/sys/fs/binfmt_misc
==> arm-image: Installing qemu-user-static in the chroot
==> arm-image: Provisioning with shell script: /tmp/packer-shell616575541
    arm-image: chroot: failed to run command ‘/bin/sh’: No such file or directory
Build 'arm-image' errored: Script exited with non-zero exit status: 127
==> Some builds didn't complete successfully and had errors:
--> arm-image: Script exited with non-zero exit status: 127
==> Builds finished but no artifacts were created.
The command "sudo ./packer build arch-raspi.json" exited with 1.
```

Any ideas? https://travis-ci.com/bcomnes/raspi-packer/builds/98912915

bcomnes, 2019-01-31:
I put together an .img src: https://github.com/bcomnes/archlinux-arm-img. I have't verified it actually works yet (I'm pretty sure it does, but still need to double check). Here is it set up to work with this plugin: https://github.com/bcomnes/raspi-packer/blob/master/arch-raspi.json#L4 but running into that `chroot: failed to run command ‘/bin/sh’: No such file or directory` error.

Perhaps I need to set up additional chroot mounts?

https://github.com/solo-io/packer-builder-arm-image/blob/master/pkg/builder/builder.go#L59

Any ideas appreciated.

JamieMagee, 2019-01-31:
@bcomnes I'm also seeing the same chmod missing error using the raspbian example. Perhaps it's an issue building on Arch Linux, see #17

bcomnes, 2019-01-31:
@JamieMagee That is a different error, and yes, perhaps an arch issue.  I'm running builds on ubuntu xenial.  I should do some sanity checks with the base examples in my environment.

yuval-k, 2019-02-04:
as i mentioned on the other issue, it seems to be related to the upstream shell provisioner.
does the `/bin/sh` binary exists in the arch linux image?

bcomnes, 2019-02-04:
Yes, but its a symlin, not sure if that affects anything: 

`lrwxrwxrwx 1 root root        4 Jan 13 14:21 sh -> bash`

bcomnes, 2019-02-04:
Path issues, eh?  Sounds like its likely part of the culprit.  @JamieMagee thank you for tracking that down!

JamieMagee, 2019-02-05:
Glad I could help. It doesn't entirely solve the issue, but it's a workaround at least.

alexandrujieanu, 2019-10-07:
Hi. How did you solve the `chroot: failed to run command ‘/bin/sh’: No such file or directory` ?

yuval-k, 2019-12-01:
closing this as solved, thanks @alexandrujieanu. added a note to the readme regarding `qemu-aarch64-static`

closed 2019-12-01 by yuval-k