solo-io/packer-plugin-arm-image #13: Fails at go build

hoshsadiq, 2018-09-24:
Seems to fail at go build. I think incompatible libraries? Perhaps upgrade or something? Below is a log.

```
    default: + go build
    default: # github.com/solo-io/packer-builder-arm-image/vendor/github.com/hashicorp/packer/packer
    default: vendor/github.com/hashicorp/packer/packer/progressbar.go:46:34: spb.BasicProgressBar.ProgressBar.SetUnits undefined (type *pb.ProgressBar has no field or method SetUnits)
    default: vendor/github.com/hashicorp/packer/packer/progressbar.go:46:44: undefined: pb.U_BYTES
    default: vendor/github.com/hashicorp/packer/packer/progressbar.go:61:5: spb.SetTotal64 undefined (type *StackableProgressBar has no field or method SetTotal64)
    default: vendor/github.com/hashicorp/packer/packer/progressbar.go:67:34: spb.BasicProgressBar.ProgressBar.Prefix undefined (type *pb.ProgressBar has no field or method Prefix)
    default: vendor/github.com/hashicorp/packer/packer/progressbar.go:93:5: cannot use new(BasicProgressBar) (type *BasicProgressBar) as type ProgressBar in assignment:
    default: 	*BasicProgressBar does not implement ProgressBar (wrong type for Finish method)
    default: 		have Finish() *pb.ProgressBar
    default: 		want Finish()
    default: vendor/github.com/hashicorp/packer/packer/progressbar.go:96:5: bpb.SetTotal64 undefined (type *BasicProgressBar has no field or method SetTotal64)
    default: vendor/github.com/hashicorp/packer/packer/progressbar.go:106:14: cannot use bpb (type *BasicProgressBar) as type ProgressBar in field value:
    default: 	*BasicProgressBar does not implement ProgressBar (wrong type for Finish method)
    default: 		have Finish() *pb.ProgressBar
    default: 		want Finish()
    default: vendor/github.com/hashicorp/packer/packer/progressbar.go:112:14: cannot use bpb (type *BasicProgressBar) as type ProgressBar in field value:
    default: 	*BasicProgressBar does not implement ProgressBar (wrong type for Finish method)
    default: 		have Finish() *pb.ProgressBar
    default: 		want Finish()
    default: # github.com/solo-io/packer-builder-arm-image/vendor/github.com/masterzen/winrm
    default: vendor/github.com/masterzen/winrm/auth.go:38:18: cannot use "github.com/solo-io/packer-builder-arm-image/vendor/github.com/masterzen/azure-sdk-for-go/core/tls".Config literal (type *"github.com/solo-io/packer-builder-arm-image/vendor/github.com/masterzen/azure-sdk-for-go/core/tls".Config) as type *"github.com/solo-io/packer-builder-arm-image/vendor/github.com/MSOpenTech/azure-sdk-for-go/core/tls".Config in field value
    default: + [[ ! -f packer-builder-arm-image ]]
    default: + echo 'Error Plugin failed to build.'
    default: + exit
    default: Error Plugin failed to build.
==> default: Running provisioner: build-image (shell)...
    default: Running: /var/folders/xy/w5lbbbz97r5c8cfxjx2n9vcxqvhpdd/T/vagrant-shell20180924-88593-w9q6gl.sh
    default: + set -e
    default: + export PACKERFILE=example.json
    default: + PACKERFILE=example.json
    default: + mkdir -p /home/vagrant/.packer.d/plugins
    default: + cp /vagrant/packer-builder-arm-image /home/vagrant/.packer.d/plugins/
    default: cp: cannot stat '/vagrant/packer-builder-arm-image'
    default: : No such file or directory
The SSH command responded with a non-zero exit status. Vagrant
assumes that this means the command failed. The output for this command
should be in the log above. Please read the output to determine what
went wrong.
```

Tried running manually, still nothing
```
 $ cd work/src/github.com/solo-io/packer-builder-arm-image/
 $ ls -la
total 92
drwxr-xr-x 8 vagrant vagrant  4096 Sep 24 22:43 .
drwxrwxr-x 3 vagrant vagrant  4096 Sep 24 22:42 ..
drwxr-xr-x 7 vagrant vagrant  4096 Sep 24 22:34 .git
-rw-r--r-- 1 vagrant vagrant    48 Sep 24 22:28 .gitignore
drwxr-xr-x 3 vagrant vagrant  4096 Sep 24 22:39 .vagrant
-rw-rw-r-- 1 vagrant vagrant 15532 Sep 24 22:43 Gopkg.lock
-rw-r--r-- 1 vagrant vagrant   943 Sep 24 22:28 Gopkg.toml
-rw-r--r-- 1 vagrant vagrant  5663 Sep 24 22:28 README.md
-rw-r--r-- 1 vagrant vagrant  1121 Sep 24 22:28 Vagrantfile
drwxr-xr-x 3 vagrant vagrant  4096 Sep 24 22:28 cmd
drwxr-xr-x 2 vagrant vagrant  4096 Sep 24 22:28 contrib
-rw-r--r-- 1 vagrant vagrant   492 Sep 24 22:28 example.json
-rw-r--r-- 1 vagrant vagrant   396 Sep 24 22:28 main.go
-rw-r--r-- 1 vagrant vagrant   376 Sep 24 22:29 micro-raspbian.iml
drwxr-xr-x 7 vagrant vagrant  4096 Sep 24 22:28 pkg
-rw-r--r-- 1 vagrant vagrant  1463 Sep 24 22:28 provision-build-env.sh
-rw-r--r-- 1 vagrant vagrant  1748 Sep 24 22:28 provision-build-image.sh
-rw-r--r-- 1 vagrant vagrant   923 Sep 24 22:28 provision-packer-builder-arm-image.sh
drwxrwxr-x 5 vagrant vagrant  4096 Sep 24 22:43 vendor
 $ dep ensure
 $ go build
# github.com/solo-io/packer-builder-arm-image/vendor/github.com/hashicorp/packer/packer
vendor/github.com/hashicorp/packer/packer/progressbar.go:46:34: spb.BasicProgressBar.ProgressBar.SetUnits undefined (type *pb.ProgressBar has no field or method SetUnits)
vendor/github.com/hashicorp/packer/packer/progressbar.go:46:44: undefined: pb.U_BYTES
vendor/github.com/hashicorp/packer/packer/progressbar.go:61:5: spb.SetTotal64 undefined (type *StackableProgressBar has no field or method SetTotal64)
vendor/github.com/hashicorp/packer/packer/progressbar.go:67:34: spb.BasicProgressBar.ProgressBar.Prefix undefined (type *pb.ProgressBar has no field or method Prefix)
vendor/github.com/hashicorp/packer/packer/progressbar.go:93:5: cannot use new(BasicProgressBar) (type *BasicProgressBar) as type ProgressBar in assignment:
	*BasicProgressBar does not implement ProgressBar (wrong type for Finish method)
		have Finish() *pb.ProgressBar
		want Finish()
vendor/github.com/hashicorp/packer/packer/progressbar.go:96:5: bpb.SetTotal64 undefined (type *BasicProgressBar has no field or method SetTotal64)
vendor/github.com/hashicorp/packer/packer/progressbar.go:106:14: cannot use bpb (type *BasicProgressBar) as type ProgressBar in field value:
	*BasicProgressBar does not implement ProgressBar (wrong type for Finish method)
		have Finish() *pb.ProgressBar
		want Finish()
vendor/github.com/hashicorp/packer/packer/progressbar.go:112:14: cannot use bpb (type *BasicProgressBar) as type ProgressBar in field value:
	*BasicProgressBar does not implement ProgressBar (wrong type for Finish method)
		have Finish() *pb.ProgressBar
		want Finish()
# github.com/solo-io/packer-builder-arm-image/vendor/github.com/masterzen/winrm
vendor/github.com/masterzen/winrm/auth.go:38:18: cannot use "github.com/solo-io/packer-builder-arm-image/vendor/github.com/masterzen/azure-sdk-for-go/core/tls".Config literal (type *"github.com/solo-io/packer-builder-arm-image/vendor/github.com/masterzen/azure-sdk-for-go/core/tls".Config) as type *"github.com/solo-io/packer-builder-arm-image/vendor/github.com/MSOpenTech/azure-sdk-for-go/core/tls".Config in field value
```

yuval-k, 2018-09-28:
Very weird!
You just did dep ensure and go build?
Will check it out soon and post back

yuval-k, 2018-09-28:
it tried it and it worked for me, to make sure it's not dep messing with us, i've pushed the vendor folder to master. can you give it a try?

yuval-k, 2018-09-28:
for record keeping, this is the relevant commit: https://github.com/solo-io/packer-builder-arm-image/commit/e32b1b5373d221789408176f2618be2b0da177d7

hannseman, 2018-10-03:
I have the same problem when running `vagrant up`. Logs: https://gist.github.com/hannseman/3febf79877240f393652be67419d7cd4

cryptobioz, 2018-11-13:
@hannseman @yuval-k Now you've added a vendor directory, I guess doing a `dep ensure` is useless. I had the same issue, removing this line from the script [provision-packer-builder-arm-image.sh](https://github.com/solo-io/packer-builder-arm-image/blob/master/provision-packer-builder-arm-image.sh#L23) fixed my issue.

commit aafcf085 (2018-11-18): no need to dep ensure

fixes #13 
thanks @cryptobioz
closed 2018-11-18 by yuval-k