solo-io/packer-plugin-arm-image #17: chmod: not found with example.json

robbert229, 2019-01-27:
**Environment:**
I am running master branch, though I am experiencing this with older versions. I am running arch linux. 

**Observed**
When I do `sudo packer build ./example.json` with the example packer file, I get the following output.

```
arm-image output will be in this color.

==> arm-image: Downloading or copying Image
    arm-image: Found already downloaded, initial checksum matched, no download needed: https://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2017-12-01/2017-11-29-raspbian-stretch-lite.zip
==> arm-image: Copying source image.
==> arm-image: Image is a zip file.
==> arm-image: Unzipping 2017-11-29-raspbian-stretch-lite.img
==> arm-image: Resizing the last partition 1073741824.
    arm-image: mappping output-arm-image/image
==> arm-image: kpartx -s -a -v output-arm-image/image
==> arm-image: partitions: [/dev/mapper/loop0p1 /dev/mapper/loop0p2]
==> arm-image: partitions: [/dev/mapper/loop0p1 /dev/mapper/loop0p2]
    arm-image: Mounting: /dev/mapper/loop0p2
    arm-image: Mounting: /dev/mapper/loop0p1
==> arm-image: Mounting additional paths within the chroot...
    arm-image: Mounting: /proc
    arm-image: Mounting: /sys
    arm-image: Mounting: /dev
    arm-image: Mounting: /dev/pts
    arm-image: Mounting: /proc/sys/fs/binfmt_misc
==> arm-image: Installing qemu-user-static in the chroot
==> arm-image: Provisioning with shell script: /tmp/packer-shell528078340
    arm-image: /bin/sh: 1: chmod: not found
    arm-image: /bin/sh: 1: /tmp/script_4790.sh: Permission denied
Build 'arm-image' errored: Script exited with non-zero exit status: 126
```

**Hunch:**
I assume that this has something to do with me running arch linux, and the chroot not getting set up correctly. Not sure, though I am digging through the source seeing if I can find why this is happening.

JamieMagee, 2019-01-31:
I'm also running Arch Linux, and also seeing the same issue

```
arm-image output will be in this color.

==> arm-image: Downloading or copying Image
    arm-image: Found already downloaded, initial checksum matched, no download needed: http://director.downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2018-11-15/2018-11-13-raspbian-stretch-lite.zip
==> arm-image: Copying source image.
==> arm-image: Image is a zip file.
==> arm-image: Unzipping 2018-11-13-raspbian-stretch-lite.img
    arm-image: mappping output-arm-image/image
==> arm-image: kpartx -s -a -v output-arm-image/image
==> arm-image: partitions: [/dev/mapper/loop0p1 /dev/mapper/loop0p2]
    arm-image: Mounting: /dev/mapper/loop0p2
    arm-image: Mounting: /dev/mapper/loop0p1
==> arm-image: Mounting additional paths within the chroot...
    arm-image: Mounting: /proc
    arm-image: Mounting: /sys
    arm-image: Mounting: /dev
    arm-image: Mounting: /dev/pts
    arm-image: Mounting: /proc/sys/fs/binfmt_misc
==> arm-image: Installing qemu-user-static in the chroot
==> arm-image: Provisioning with shell script: /tmp/packer-shell329101979
    arm-image: /bin/sh: 1: chmod: not found
    arm-image: /bin/sh: 1: /tmp/script_1618.sh: Permission denied
```

JamieMagee, 2019-02-01:
@yuval-k Any suggestions?

yuval-k, 2019-02-04:
this seems to be part of the shell provisioner in upstream packer:

https://github.com/hashicorp/packer/blob/14aa2afbb12a6f46164f199136d4b23da13709af/provisioner/shell/provisioner.go#L116

I think this error means that `chmod` is not found inside the image. can you confirm that?

corysm1th, 2019-02-04:
Not using example.json, but getting a slightly different error at the same stage of the build.
`==> arm-image: Mounting additional paths within the chroot...
    arm-image: Mounting: /proc
    arm-image: Mounting: /sys
    arm-image: Mounting: /dev
    arm-image: Mounting: /dev/pts
    arm-image: Mounting: /proc/sys/fs/binfmt_misc
==> arm-image: Installing qemu-user-static in the chroot
==> arm-image: Pausing before the next provisioner . Press enter to continue. ==> arm-image: Provisioning with shell script: /tmp/packer-shell227276521
    arm-image: chroot: failed to run command ‘/bin/sh’: No such file or directory`

Running on Ubuntu 18.04 from user account with `sudo packer...`

JamieMagee, 2019-02-04:
@yuval-k I think `$PATH` might not be set correctly. Setting `"execute_command": "/bin/chmod +x {{ .Path }}; {{ .Vars }} {{ .Path }}",` in the shell provisioner makes it work.

yuval-k, 2019-02-06:
thanks for the hint. as you can see here https://github.com/solo-io/packer-builder-arm-image/blob/2a33747c943b22b1e6642a6f226f1ab88f81fb42/pkg/builder/communicator.go#L30
the command is wrapped in a chroot before being executed... 
is there anything in archlinux that might strip the path env when doing a chroot?

JamieMagee, 2019-02-07:
I haven't had a chance to look fully, but [here's the Arch wiki page for `chroot`](https://wiki.archlinux.org/index.php/chroot) and [here's the code for `arch-chroot`](https://git.archlinux.org/arch-install-scripts.git/tree/arch-chroot.in) (normally used for installing arch and building packages)