solo-io/packer-plugin-arm-image #18: packerlicious support for packer-builder-arm-image plugin

mayn, 2019-01-31:
Hi project maintainers! 
Members from your community have added partial support for your plugin to [packerlicious](https://github.com/mayn/packerlicious) via https://github.com/mayn/packerlicious/pull/164 .
One of the challenges preventing full support was with undocumented attributes.

If you are interested in expanding packerlicious supports for your plugin. 
Please make a pr with the additional attributes.
 
I could use your help.
Thanks!

yuval-k, 2019-02-04:
do you have an example for such pr?
I'm not sure what do you mean exactly.

mayn, 2019-02-06:
hi @yuval-k . here is the properties defined for your plugin https://github.com/mayn/packerlicious/blob/master/src/packerlicious/community/builder.py#L25

Your docs indicate that additional properties are supported, so they would need to be added to that mapping.

hoshsadiq, 2019-09-24:
@mayn I'm not a packerlicious user, however, from reading the code, these are all the fields I can see that are possible (note, you want the mapstructure value):

- https://github.com/hashicorp/packer/blob/1f79b430eee6d87281f1dc207386e02c7033e80a/common/packer_config.go
- https://github.com/hashicorp/packer/blob/f0461da1a1a6234c934ecacf1c16a7458d1a088b/common/iso_config.go#L71-L109
- https://github.com/solo-io/packer-builder-arm-image/blob/6341b2160c5bce1973a1b903726422e45c915d6d/pkg/builder/builder.go#L41-L71

Hope that helps