solo-io/packer-plugin-arm-image #22: Failed to fetch dependencies

gianlazz, 2019-02-20:
Wondering if anyone can shed some light on this problem? Is it just a matter of needing to change the dependency mirrors?

This was my result from running `vagrant up`
```
    default: ==> arm-image: Provisioning with shell script: /tmp/packer-shell014610762
    default:     arm-image: Err:1 http://archive.raspberrypi.org/debian stretch InRelease
    default:     arm-image:   Temporary failure resolving 'archive.raspberrypi.org'
    default:     arm-image: Err:2 http://mirrordirector.raspbian.org/raspbian stretch InRelease
    default:     arm-image:   Temporary failure resolving 'mirrordirector.raspbian.org'
    default:     arm-image: Reading package lists...
    default:     arm-image: W: Failed to fetch http://mirrordirector.raspbian.org/raspbian/dists/stretch/InRelease  Temporary failure resolving 'mirrordirector.raspbian.org'
    default:     arm-image: W: Failed to fetch http://archive.raspberrypi.org/debian/dists/stretch/InRelease  Temporary failure resolving 'archive.raspberrypi.org'
    default:     arm-image: W: Some index files failed to download. They have been ignored, or old ones used instead.
    default:     arm-image: Reading package lists...
    default:     arm-image: Building dependency tree...
    default:     arm-image: Reading state information...
    default:     arm-image: The following additional packages will be installed:
    default:     arm-image:   golang-1.7 golang-1.7-doc golang-1.7-go golang-1.7-src golang-doc golang-go
    default:     arm-image:   golang-src
    default:     arm-image: Suggested packages:
    default:     arm-image:   bzr git mercurial subversion
    default:     arm-image: The following NEW packages will be installed:
    default:     arm-image:   golang golang-1.7 golang-1.7-doc golang-1.7-go golang-1.7-src golang-doc
    default:     arm-image:   golang-go golang-src
    default:     arm-image: 0 upgraded, 8 newly installed, 0 to remove and 0 not upgraded.
    default:     arm-image: Need to get 27.5 MB of archives.
    default:     arm-image: After this operation, 144 MB of additional disk space will be used.
    default:     arm-image: Err:1 http://mirrordirector.raspbian.org/raspbian stretch/main armhf golang-1.7-src armhf 1.7.4-2+rpi1
    default:     arm-image:   Temporary failure resolving 'mirrordirector.raspbian.org'
    default:     arm-image: Err:2 http://mirrordirector.raspbian.org/raspbian stretch/main armhf golang-1.7-go armhf 1.7.4-2+rpi1
    default:     arm-image:   Temporary failure resolving 'mirrordirector.raspbian.org'
    default:     arm-image: Err:3 http://mirrordirector.raspbian.org/raspbian stretch/main armhf golang-1.7-doc all 1.7.4-2+rpi1
    default:     arm-image:   Temporary failure resolving 'mirrordirector.raspbian.org'
    default:     arm-image: Err:4 http://mirrordirector.raspbian.org/raspbian stretch/main armhf golang-1.7 all 1.7.4-2+rpi1
    default:     arm-image:   Temporary failure resolving 'mirrordirector.raspbian.org'
    default:     arm-image: Err:5 http://mirrordirector.raspbian.org/raspbian stretch/main armhf golang-src armhf 2:1.7~5
    default:     arm-image:   Temporary failure resolving 'mirrordirector.raspbian.org'
    default:     arm-image: Err:6 http://mirrordirector.raspbian.org/raspbian stretch/main armhf golang-go armhf 2:1.7~5
    default:     arm-image:   Temporary failure resolving 'mirrordirector.raspbian.org'
    default:     arm-image: Err:7 http://mirrordirector.raspbian.org/raspbian stretch/main armhf golang-doc all 2:1.7~5
    default:     arm-image:   Temporary failure resolving 'mirrordirector.raspbian.org'
    default:     arm-image: Err:8 http://mirrordirector.raspbian.org/raspbian stretch/main armhf golang armhf 2:1.7~5
    default:     arm-image:   Temporary failure resolving 'mirrordirector.raspbian.org'
    default:     arm-image: E: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/g/golang-1.7/golang-1.7-src_1.7.4-2+rpi1_armhf.deb  Temporary failure resolving 'mirrordirector.raspbian.org'
    default:     arm-image: E: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/g/golang-1.7/golang-1.7-go_1.7.4-2+rpi1_armhf.deb  Temporary failure resolving 'mirrordirector.raspbian.org'
    default:     arm-image: E: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/g/golang-1.7/golang-1.7-doc_1.7.4-2+rpi1_all.deb  Temporary failure resolving 'mirrordirector.raspbian.org'
    default:     arm-image: E: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/g/golang-1.7/golang-1.7_1.7.4-2+rpi1_all.deb  Temporary failure resolving 'mirrordirector.raspbian.org'
    default:     arm-image: E: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/g/golang-defaults/golang-src_1.7~5_armhf.deb  Temporary failure resolving 'mirrordirector.raspbian.org'
    default:     arm-image: E: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/g/golang-defaults/golang-go_1.7~5_armhf.deb  Temporary failure resolving 'mirrordirector.raspbian.org'
    default:     arm-image: E: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/g/golang-defaults/golang-doc_1.7~5_all.deb  Temporary failure resolving 'mirrordirector.raspbian.org'
    default:     arm-image: E: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/g/golang-defaults/golang_1.7~5_armhf.deb  Temporary failure resolving 'mirrordirector.raspbian.org'
    default:     arm-image: E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
    default: Build 'arm-image' errored: Script exited with non-zero exit status: 100
    default: ==> Some builds didn't complete successfully and had errors:
    default: --> arm-image: Script exited with non-zero exit status: 100
    default:
    default: ==> Builds finished but no artifacts were created.
    default: ++ grep -Po '(?<=Build '\'').*(?='\'' finished.)' /tmp/tmp.xPTe2JFnms
    default: + BUILD_NAME=
The SSH command responded with a non-zero exit status. Vagrant
assumes that this means the command failed. The output for this command
should be in the log above. Please read the output to determine what
went wrong.
```

gianlazz, 2019-02-21:
Actually the mirrors seem to resolve just fine on my machine but rather that the emulated pi image doesn't seem to be able to connect to the internet.

yuval-k, 2019-02-22:
this builder works by creating a chroot environment - is there any reason for chroot environment on your machine to no have network access?

meringu, 2019-07-12:
I'm getting a similar issue in Ubuntu pre-installed for Pi. Ubuntu recommend mounting the resolv.conf  in these docs: https://help.ubuntu.com/community/BasicChroot Is this something that could help in this case?

meringu, 2019-07-13:
Looks like my problem was due to the `/etc/resolv.conf` file being symlinked to `/run/systemd/resolve/stub-resolv.conf`, however the `/run/systemd` was missing entirely.

When my raspberry PI is booted, `/run/systemd` is present, but not when using this packer builder for some reason.

I mounted the `/run/systemd` folder with the `chroot_mounts` option, by adding `["bind", "/run/systemd", "/run/systemd"]` to the default options.

After this, my `resolv.conf` was linked up again, and DNS was working.

@gianlazz, I'd recommend checking out the state of your `/etc/resolv.conf` file, as it could be the reason your DNS resolution isn't working.

yuval-k, 2019-12-01:
closing as answered. added `additional_chroot_mounts` to make this use case easier.

closed 2019-12-01 by yuval-k