solo-io/packer-plugin-arm-image #23: Contribute upstream?

JamieMagee, 2019-03-04:
I think this is a very useful builder for packer, and I think that contributing it upstream would allow more people to find and use it.

yuval-k, 2019-03-06:
Do you know what that entails? I'm not sure I'll have the bandwidth to do so in the near term, as we are currently putting a lot of focus on [gloo](https://github.com/solo-io/gloo)

JamieMagee, 2019-03-09:
I'm not 100% sure, as I've not done it before, however they have a [contributing guide here](https://github.com/hashicorp/packer/blob/master/.github/CONTRIBUTING.md), and [here's a PR](https://github.com/hashicorp/packer/pull/7294) adding a builder for the HyperOne cloud platform. Seems straightforward in terms of engineering

I'm not so sure on the legal side of things. `packer` is licensed under the [Mozilla Public License 2.0](https://github.com/hashicorp/packer/blob/master/LICENSE) and they require contributors to sign a CLA. as `packer-builder-arm-image` is currently licensed under the Apache License 2.0. I don't believe I can do this, as I am not the copyright owner of the code. If it's okay with you, I'll open an issue upstream asking for clarification on this matter.