solo-io/packer-plugin-arm-image #26: Trying to build alpine

hoshsadiq, 2019-05-11:
Hi

I'm trying to build an image based on alpine but it complains about partition issues:
```
vagrant@ubuntu-xenial:/vagrant$ ./provision-build-image.sh
+ set -e
+ export PACKERFILE=example.json
+ PACKERFILE=example.json
+ mkdir -p /home/vagrant/.packer.d/plugins
+ cp /vagrant/packer-builder-arm-image /home/vagrant/.packer.d/plugins/
+ [[ ! -f /home/vagrant/.packer.d/plugins/packer-builder-arm-image ]]
+ echo 'Attempting to build image'
Attempting to build image
++ mktemp
+ PACKER_LOG=/tmp/tmp.PVVUzssbLo
+ [[ -f /vagrant/example.json ]]
+ sudo packer build /vagrant/example.json
+ tee /tmp/tmp.PVVUzssbLo
arm-image output will be in this color.

==> arm-image: Downloading or copying Image
    arm-image: Downloading or copying: file://alpine-rpi-3.9.4-aarch64.tar.gz
==> arm-image: Copying source image.
    arm-image: mappping output-arm-image/image
==> arm-image: kpartx -s -a -v output-arm-image/image
==> arm-image: partitions: []
==> arm-image: error different of partitions than expected 0
Build 'arm-image' errored: step canceled or halted

==> Some builds didn't complete successfully and had errors:
--> arm-image: step canceled or halted

==> Builds finished but no artifacts were created.
++ grep -Po '(?<=Build '\'').*(?='\'' finished.)' /tmp/tmp.PVVUzssbLo
+ BUILD_NAME=
```

My packer json looks like this (the tar.gz is available [here](http://dl-cdn.alpinelinux.org/alpine/v3.9/releases/aarch64/alpine-rpi-3.9.4-aarch64.tar.gz)):
```json
{
  "builders": [
    {
      "type": "arm-image",
      "iso_url": "file://alpine-rpi-3.9.4-aarch64.tar.gz",
      "iso_checksum_type": "sha256",
      "iso_checksum": "76af2ee59f5865dedd2c940402d3590937236f29ef1aaf52d9f4d11df8afc79a",
      "image_type": "raspberrypi",
      "image_mounts": ["/boot", "/"]
    }
  ],
  "provisioners": [
    {
      "type": "shell",
      "inline": [
        "cat /etc/os-release"
      ]
    }
  ]
}
```

hoshsadiq, 2019-05-11:
I've just tried building with the latest raspbian late as well, getting the same error:

```
vagrant@ubuntu-xenial:/vagrant$ ./provision-build-image.sh
+ set -e
+ export PACKERFILE=example.json
+ PACKERFILE=example.json
+ mkdir -p /home/vagrant/.packer.d/plugins
+ cp /vagrant/packer-builder-arm-image /home/vagrant/.packer.d/plugins/
+ [[ ! -f /home/vagrant/.packer.d/plugins/packer-builder-arm-image ]]
+ echo 'Attempting to build image'
Attempting to build image
++ mktemp
+ PACKER_LOG=/tmp/tmp.lGiKA4R7nE
+ [[ -f /vagrant/example.json ]]
+ sudo packer build /vagrant/example.json
+ tee /tmp/tmp.lGiKA4R7nE
arm-image output will be in this color.

==> arm-image: Downloading or copying Image
    arm-image: Downloading or copying: https://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2019-04-09/2019-04-08-raspbian-stretch-lite.zip
    arm-image: Download progress: 62%
    arm-image: Download progress: 100%
==> arm-image: Copying source image.
==> arm-image: Image is a zip file.
==> arm-image: Unzipping 2019-04-08-raspbian-stretch-lite.img
    arm-image: Speed:   70.62 MB/s
    arm-image: Progress: 58.74%
    arm-image: mappping output-arm-image/image
==> arm-image: kpartx -s -a -v output-arm-image/image
==> arm-image: partitions: []
==> arm-image: error different of partitions than expected 0
Build 'arm-image' errored: step canceled or halted

==> Some builds didn't complete successfully and had errors:
--> arm-image: step canceled or halted

==> Builds finished but no artifacts were created.
++ grep -Po '(?<=Build '\'').*(?='\'' finished.)' /tmp/tmp.lGiKA4R7nE
+ BUILD_NAME=
```

yuval-k, 2019-06-06:
It seems (from here https://wiki.alpinelinux.org/wiki/Raspberry_Pi#Preparation) that the alpine tar.gz is not an image disk; i would do the preparation section on a file instead of an sd card, and then use that file as the input for this program.

hoshsadiq, 2019-06-07:
Do you have any good resources on how to do that? I've been able to create an ISO from the tar.gz but been able to separate the boot and the root partition. Either way, packer doesn't accept that either.

yuval-k, 2019-06-09:
steps 1-4 here: https://lihongxc.blogspot.com/2009/03/create-root-file-system-image.html are roughly what you need

hoshsadiq, 2019-06-17:
Thanks for that. Based on your link and some other research, I've come up with the below script to create rootfs image:
```bash
#!/usr/bin/env bash

set -euxo pipefail

umount /tmp/mnt || true
losetup --detach /dev/loop0 || true
rm -rf root.img || true

# create an empty image
dd if=/dev/zero of=root.img seek=256M bs=1 count=0

{
    echo o      # Create a new empty DOS partition table

    # set up partition 1
    echo n      # Add a new partition
    echo p      # Primary partition
    echo 1      # Partition 1
    echo        # use default for first sector
    echo +64M   # Last sector
    echo t      # set partition type
    echo c      # set partition type to W95 FAT32 (LBA)
    echo a      # set partition 1 as bootable/active

    # now we create partition 2
    echo n      # new partition
    echo p      # primary partition
    echo 2      # partion number 2
    echo        # default, start immediately after preceding partition
    echo        # default, extend partition to end of disk

    # print the in-memory partition table
    echo p

    # save and quit
    echo w
    echo q
} | fdisk root.img

losetup --show -P /dev/loop0 root.img
mkfs.vfat /dev/loop0p1
mkfs.ext4 /dev/loop0p2
losetup --detach /dev/loop0

rm -rf /tmp/mnt || true
mkdir -p /tmp/mnt

startSector="$(sfdisk -J ./root.img | jq '.partitiontable.partitions[] | select(.type == "c") | .start')"
offset="$(( $startSector * 512 ))"
mount -t vfat -o loop -o "offset=$offset" root.img /tmp/mnt

curl -fsSL http://dl-cdn.alpinelinux.org/alpine/v3.9/releases/aarch64/alpine-rpi-3.9.4-aarch64.tar.gz -o- | tar -xzv --no-same-owner -C /tmp/mnt --file -

umount /tmp/mnt

sum="$(sha256sum root.img  | awk '{print $1}')"
alpineJson="$(jq --arg sha256sum "$sum" '. | .builders[0].iso_checksum = $sha256sum' packer-alpine.json)"
echo "$alpineJson" > packer-alpine.json
```

With the image generated by above script I get the following result:
```bash
root@ubuntu-xenial:/vagrant# ./provision-build-image.sh 
+ set -e
+ export PACKERFILE=packer-alpine.json
+ PACKERFILE=packer-alpine.json
+ mkdir -p /home/vagrant/.packer.d/plugins
+ cp /vagrant/packer-builder-arm-image /home/vagrant/.packer.d/plugins/
+ [[ ! -f /home/vagrant/.packer.d/plugins/packer-builder-arm-image ]]
+ echo 'Attempting to build image'
Attempting to build image
++ mktemp
+ PACKER_LOG=/tmp/tmp.wd0Bh72AHF
+ [[ -f /vagrant/packer-alpine.json ]]
+ tee /tmp/tmp.wd0Bh72AHF
+ sudo packer build -debug /vagrant/packer-alpine.json
Debug mode enabled. Builds will not be parallelized.
arm-image output will be in this color.

==> arm-image: Downloading or copying Image
    arm-image: Downloading or copying: file://root.img
==> arm-image: Copying source image.
    arm-image: mappping output-arm-image/image
==> arm-image: kpartx -s -a -v output-arm-image/image
==> arm-image: partitions: [/dev/mapper/loop0p1 /dev/mapper/loop0p2]
    arm-image: Mounting: /dev/mapper/loop0p2
    arm-image: Mounting: /dev/mapper/loop0p1
==> arm-image: Error executing command 'mount /dev/mapper/loop0p1 /tmp/517219996/boot': exit status 32
==> arm-image: Stderr: mount: mount point /tmp/517219996/boot does not exist
==> arm-image:
Build 'arm-image' errored: Error executing command 'mount /dev/mapper/loop0p1 /tmp/517219996/boot': exit status 32
Stderr: mount: mount point /tmp/517219996/boot does not exist

==> Some builds didn't complete successfully and had errors:
--> arm-image: Error executing command 'mount /dev/mapper/loop0p1 /tmp/517219996/boot': exit status 32
Stderr: mount: mount point /tmp/517219996/boot does not exist

==> Builds finished but no artifacts were created.
++ grep -Po '(?<=Build '\'').*(?='\'' finished.)' /tmp/tmp.wd0Bh72AHF
+ BUILD_NAME=
```

yuval-k, 2019-06-19:
To fix the error above, I think you need to add `mkdir /tmp/mnt/boot` right before `umount /tmp/mnt` in your script.

hoshsadiq, 2019-06-20:
That dir already exists. For good measure I mounted the image cleanly and check, and the boot dir exists.

Thanks for all the help by the way! Much appreciated :)

```
root@ubuntu-xenial:/vagrant# ls -la /tmp/mnt/
total 3028
drwxr-xr-x 5 root root   16384 May  9 20:50 .
drwxrwxrwt 8 root root    4096 Jun 20 09:46 ..
-rwxr-xr-x 1 root root      24 May  9 20:50 .alpine-release
drwxr-xr-x 3 root root    2048 May  9 20:50 apks
-rwxr-xr-x 1 root root   25505 May  9 11:00 bcm2710-rpi-3-b.dtb
-rwxr-xr-x 1 root root   26124 May  9 11:00 bcm2710-rpi-3-b-plus.dtb
-rwxr-xr-x 1 root root   24277 May  9 11:00 bcm2710-rpi-cm3.dtb
-rwxr-xr-x 1 root root   19392 May  9 11:00 bcm2837-rpi-3-b.dtb
-rwxr-xr-x 1 root root   20181 May  9 11:00 bcm2837-rpi-3-b-plus.dtb
drwxr-xr-x 2 root root    2048 May  9 20:50 boot
-rwxr-xr-x 1 root root   52296 May  9 19:59 bootcode.bin
-rwxr-xr-x 1 root root      81 May  9 20:49 cmdline.txt
-rwxr-xr-x 1 root root     231 May  9 20:49 config.txt
-rwxr-xr-x 1 root root    6701 May  9 19:59 fixup.dat
drwxr-xr-x 2 root root   14336 May  9 20:50 overlays
-rwxr-xr-x 1 root root 2873444 May  9 19:59 start.elf
root@ubuntu-xenial:/vagrant# ls -la /tmp/mnt/boot
total 44236
drwxr-xr-x 2 root root     2048 May  9 20:50 .
drwxr-xr-x 5 root root    16384 May  9 20:50 ..
-rwxr-xr-x 1 root root   153783 May  9 20:50 config-rpi
-rwxr-xr-x 1 root root  4637528 May  9 20:50 initramfs-rpi
-rwxr-xr-x 1 root root 25378816 May  9 20:50 modloop-rpi
-rwxr-xr-x 1 root root  2675912 May  9 20:50 System.map-rpi
-rwxr-xr-x 1 root root 12427776 May  9 20:50 vmlinuz-rpi
r
root@ubuntu-xenial:/vagrant# startSector="$(sfdisk -J ./root.img | jq '.partitiontable.partitions[] | select(.type == "c") | .start')"
root@ubuntu-xenial:/vagrant# offset="$(( $startSector * 512 ))"
root@ubuntu-xenial:/vagrant# mount -t vfat -o loop -o "offset=$offset" root.img /mnt
root@ubuntu-xenial:/vagrant# ls -la /mnt/
total 3028
drwxr-xr-x  5 root root   16384 Jan  1  1970 .
drwxr-xr-x 24 root root    4096 Jun 20 09:42 ..
-rwxr-xr-x  1 root root      24 May  9 20:50 .alpine-release
drwxr-xr-x  3 root root    2048 May  9 20:50 apks
-rwxr-xr-x  1 root root   25505 May  9 11:00 bcm2710-rpi-3-b.dtb
-rwxr-xr-x  1 root root   26124 May  9 11:00 bcm2710-rpi-3-b-plus.dtb
-rwxr-xr-x  1 root root   24277 May  9 11:00 bcm2710-rpi-cm3.dtb
-rwxr-xr-x  1 root root   19392 May  9 11:00 bcm2837-rpi-3-b.dtb
-rwxr-xr-x  1 root root   20181 May  9 11:00 bcm2837-rpi-3-b-plus.dtb
drwxr-xr-x  2 root root    2048 May  9 20:50 boot
-rwxr-xr-x  1 root root   52296 May  9 19:59 bootcode.bin
-rwxr-xr-x  1 root root      81 May  9 20:49 cmdline.txt
-rwxr-xr-x  1 root root     231 May  9 20:49 config.txt
-rwxr-xr-x  1 root root    6701 May  9 19:59 fixup.dat
drwxr-xr-x  2 root root   14336 May  9 20:50 overlays
-rwxr-xr-x  1 root root 2873444 May  9 19:59 start.elf
root@ubuntu-xenial:/vagrant# ls -la /mnt/boot
total 44236
drwxr-xr-x 2 root root     2048 May  9 20:50 .
drwxr-xr-x 5 root root    16384 Jan  1  1970 ..
-rwxr-xr-x 1 root root   153783 May  9 20:50 config-rpi
-rwxr-xr-x 1 root root  4637528 May  9 20:50 initramfs-rpi
-rwxr-xr-x 1 root root 25378816 May  9 20:50 modloop-rpi
-rwxr-xr-x 1 root root  2675912 May  9 20:50 System.map-rpi
-rwxr-xr-x 1 root root 12427776 May  9 20:50 vmlinuz-rpi
r
```

yuval-k, 2019-06-23:
Ok i think i see what's going, i was running your script and noticed you never used the root folder.. try this script (replace the part after fdisk):
```
losetup --show -P /dev/loop0 root.img
mkfs.vfat /dev/loop0p1
mkfs.ext4 /dev/loop0p2

rm -rf /tmp/mnt || true
mkdir -p /tmp/mnt

mount -t ext4 /dev/loop0p2 /tmp/mnt
mkdir -p /tmp/mnt/boot
mount -t vfat /dev/loop0p1 /tmp/mnt/boot

curl -fsSL http://dl-cdn.alpinelinux.org/alpine/v3.9/releases/aarch64/alpine-rpi-3.9.4-aarch64.tar.gz -o- | tar -xzv --no-same-owner -C /tmp/mnt --file -

umount /tmp/mnt/boot
umount /tmp/mnt/

sum="$(sha256sum root.img  | awk '{print $1}')"
alpineJson="$(jq --arg sha256sum "$sum" '. | .builders[0].iso_checksum = $sha256sum' packer-alpine.json)"
echo "$alpineJson" > packer-alpine.json

losetup --detach /dev/loop0
```

yuval-k, 2019-06-23:
this is assuming alpine supports a separate partition for root - which i'm not sure if true, but i'll leave that to you..

yuval-k, 2019-06-23:
the reason for failure is that the plugin tries to mount the root partition first, and it was empty as all the files were in the boot partition.

hoshsadiq, 2019-06-24:
> this is assuming alpine supports a separate partition for root - which i'm not sure if true, but i'll leave that to you..

It does, however, by default it's meant to be running from memory/boot without a root partition. The idea is to set it up after first boot and setup normal `sys` mode installation (that is, it'll be the normal setup where things can get saves etc). I'm trying to do this setup using packer and this plugin.

Your suggested solution got it further but now it complains about not being to add `/usr/bin/qemu-arm-static` (see below).

I think this is to do with Alpine's rootfs not being conventional due to it running in non-sysmode. `/usr/bin` wouldn't exist.

```
root@ubuntu-xenial:/vagrant# ./provision-build-image.sh
+ set -e
+ export PACKERFILE=packer-alpine.json
+ PACKERFILE=packer-alpine.json
+ mkdir -p /home/vagrant/.packer.d/plugins
+ cp /vagrant/packer-builder-arm-image /home/vagrant/.packer.d/plugins/
+ [[ ! -f /home/vagrant/.packer.d/plugins/packer-builder-arm-image ]]
+ echo 'Attempting to build image'
Attempting to build image
++ mktemp
+ PACKER_LOG=/tmp/tmp.GgLvAE4DWV
+ [[ -f /vagrant/packer-alpine.json ]]
+ sudo packer build -debug /vagrant/packer-alpine.json
+ tee /tmp/tmp.GgLvAE4DWV
Debug mode enabled. Builds will not be parallelized.
arm-image output will be in this color.

==> arm-image: Downloading or copying Image
    arm-image: Downloading or copying: file://root.img
==> arm-image: Copying source image.
    arm-image: mappping output-arm-image/image
==> arm-image: kpartx -s -a -v output-arm-image/image
==> arm-image: partitions: [/dev/mapper/loop0p1 /dev/mapper/loop0p2]
    arm-image: Mounting: /dev/mapper/loop0p2
    arm-image: Mounting: /dev/mapper/loop0p1
==> arm-image: Mounting additional paths within the chroot...
    arm-image: Mounting: /proc
    arm-image: Mounting: /sys
    arm-image: Mounting: /dev
    arm-image: Mounting: /dev/pts
    arm-image: Mounting: /proc/sys/fs/binfmt_misc
==> arm-image: Installing qemu-user-static in the chroot
==> arm-image: Error executing command 'cp /usr/bin/qemu-arm-static /tmp/209261126/usr/bin/qemu-arm-static': exit status 1
==> arm-image: Stderr: cp: cannot create regular file '/tmp/209261126/usr/bin/qemu-arm-static': No such file or directory
==> arm-image:
Build 'arm-image' errored: Error executing command 'cp /usr/bin/qemu-arm-static /tmp/209261126/usr/bin/qemu-arm-static': exit status 1
Stderr: cp: cannot create regular file '/tmp/209261126/usr/bin/qemu-arm-static': No such file or directory

==> Some builds didn't complete successfully and had errors:
--> arm-image: Error executing command 'cp /usr/bin/qemu-arm-static /tmp/209261126/usr/bin/qemu-arm-static': exit status 1
Stderr: cp: cannot create regular file '/tmp/209261126/usr/bin/qemu-arm-static': No such file or directory

==> Builds finished but no artifacts were created.
++ grep -Po '(?<=Build '\'').*(?='\'' finished.)' /tmp/tmp.GgLvAE4DWV
+ BUILD_NAME=
```

yuval-k, 2019-12-01:
Hi!
I believe this should be fixed in the latest version (v0.1.3); I made a change to just copy it to the root folder instead of `/usr/bin` for max comparability.

closing as resolved, feel free to re-open if needed

closed 2019-12-01 by yuval-k

hoshsadiq, 2019-12-01:
I guess this doesn't actually matter. Reason is alpine uses initramfs and only boots up from there. Changing the location probably won't fix it. Thanks anyway though!