solo-io/packer-plugin-arm-image #28: Build 'arm-image' errored: Image download failed.

gianarb, 2019-07-25:
```json
  "builders": [
    {
      "type": "arm-image",
      "iso_url": "https://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2019-07-12/2019-07-10-raspbian-buster-lite.zip",
      "iso_checksum_type": "sha256",
      "iso_checksum": "9e5cf24ce483bb96e7736ea75ca422e3560e7b455eee63dd28f66fa1825db70e",
      "last_partition_extra_size": 1073741824
    }
  ],
```
this is my configuration but when I run `packer build` the download doesn't even start. I don't know what I am missing:

```bash
$ sudo -u root HOME=/home/myuser WIFI_NAME= WIFI_PASS= ./build.sh
Debug mode enabled. Builds will not be parallelized.
arm-image output will be in this color.

==> arm-image: Downloading or copying Image
    arm-image: Downloading or copying: https://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2019-07-12/2019-07-10-raspbian-buster-lite.zip
    arm-image: Error downloading: open : no such file or directory
==> arm-image: Image download failed.
Build 'arm-image' errored: Image download failed.

==> Some builds didn't complete successfully and had errors:
--> arm-image: Image download failed.

==> Builds finished but no artifacts were created.
```

```
$ packer -v
1.4.2
```
I compiled packer-builder-arm-image cloning the master and I am on ArchLinux. What else I can do? Thanks a lot

matt-deboer, 2019-07-29:
It looks to be related to this issue: https://github.com/hashicorp/packer/issues/7622#issuecomment-490486542
I can confirm that downgrading packer to 1.3.5 (as they mentioned) gets me past this issue.

gianarb, 2019-07-30:
Sweet, it looks like a deprecation from their side. Thanks a lot

jangrewe, 2019-11-26:
Is there any chance to get this plugin fixed? I'm not good at Go (or anything), but i guess one of the devs should be able to implement these changes: https://github.com/hashicorp/packer/pull/7583/files

yuval-k, 2019-11-27:
yea now that kubecon is over i should have some time during the weekend to take a look. thanks for the reference

yuval-k, 2019-11-30:
code updated to packer 1.4
closing as resolved

closed 2019-11-30 by yuval-k