solo-io/packer-plugin-arm-image #29: Failed to initialize build 'arm-image': error initializing builder 'arm-image': fork/exec /usr/bin/packer-builder-arm-image: no such file or directory

matt-deboer, 2019-07-29:
```Failed to initialize build 'arm-image': error initializing builder 'arm-image': fork/exec /usr/bin/packer-builder-arm-image: no such file or directory```

Any ideas why I would run into the following error?

Checking the obvious shows that the file does in fact exist with sensible flags:
```
$ ls -lah /usr/bin/packer-builder-arm-image 
-rwxr-xr-x 1 root root 14M Jul 29 10:01 /usr/bin/packer-builder-arm-image

$ file /usr/bin/packer-builder-arm-image 
/usr/bin/packer-builder-arm-image: ELF 64-bit LSB executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib/ld-, not stripped
```

**Note:** I downloaded the executable directly from the `v0.1.0` release

Complete build log is here:
```
2019/07/29 10:12:19 [INFO] Packer version: 1.4.2
2019/07/29 10:12:19 Packer Target OS/Arch: linux amd64
2019/07/29 10:12:19 Built with Go Version: go1.12.6
2019/07/29 10:12:19 [DEBUG] Discovered plugin: arm-image = /usr/bin/packer-builder-arm-image
2019/07/29 10:12:19 Detected home directory from env var: /home/dummy
2019/07/29 10:12:19 Using internal plugin for amazon-chroot
2019/07/29 10:12:19 Using internal plugin for amazon-ebssurrogate
2019/07/29 10:12:19 Using internal plugin for hyperv-iso
2019/07/29 10:12:19 Using internal plugin for ncloud
2019/07/29 10:12:19 Using internal plugin for profitbricks
2019/07/29 10:12:19 Using internal plugin for scaleway
2019/07/29 10:12:19 Using internal plugin for tencentcloud-cvm
2019/07/29 10:12:19 Using internal plugin for vmware-vmx
2019/07/29 10:12:19 Using internal plugin for amazon-ebsvolume
2019/07/29 10:12:19 Using internal plugin for amazon-instance
2019/07/29 10:12:19 Using internal plugin for digitalocean
2019/07/29 10:12:19 Using internal plugin for lxd
2019/07/29 10:12:19 Using internal plugin for oneandone
2019/07/29 10:12:19 Using internal plugin for virtualbox-iso
2019/07/29 10:12:19 Using internal plugin for hcloud
2019/07/29 10:12:19 Using internal plugin for hyperv-vmcx
2019/07/29 10:12:19 Using internal plugin for lxc
2019/07/29 10:12:19 Using internal plugin for qemu
2019/07/29 10:12:19 Using internal plugin for alicloud-ecs
2019/07/29 10:12:19 Using internal plugin for amazon-ebs
2019/07/29 10:12:19 Using internal plugin for proxmox
2019/07/29 10:12:19 Using internal plugin for virtualbox-ovf
2019/07/29 10:12:19 Using internal plugin for googlecompute
2019/07/29 10:12:19 Using internal plugin for hyperone
2019/07/29 10:12:19 Using internal plugin for null
2019/07/29 10:12:19 Using internal plugin for oracle-oci
2019/07/29 10:12:19 Using internal plugin for parallels-iso
2019/07/29 10:12:19 Using internal plugin for parallels-pvm
2019/07/29 10:12:19 Using internal plugin for azure-arm
2019/07/29 10:12:19 Using internal plugin for file
2019/07/29 10:12:19 Using internal plugin for linode
2019/07/29 10:12:19 Using internal plugin for yandex
2019/07/29 10:12:19 Using internal plugin for cloudstack
2019/07/29 10:12:19 Using internal plugin for docker
2019/07/29 10:12:19 Using internal plugin for vmware-iso
2019/07/29 10:12:19 Using internal plugin for vagrant
2019/07/29 10:12:19 Using internal plugin for openstack
2019/07/29 10:12:19 Using internal plugin for oracle-classic
2019/07/29 10:12:19 Using internal plugin for triton
2019/07/29 10:12:19 Using internal plugin for ansible
2019/07/29 10:12:19 Using internal plugin for chef-solo
2019/07/29 10:12:19 Using internal plugin for file
2019/07/29 10:12:19 Using internal plugin for inspec
2019/07/29 10:12:19 Using internal plugin for shell
2019/07/29 10:12:19 Using internal plugin for shell-local
2019/07/29 10:12:19 Using internal plugin for breakpoint
2019/07/29 10:12:19 Using internal plugin for chef-client
2019/07/29 10:12:19 Using internal plugin for puppet-masterless
[... 24 lines cut ...]
2019/07/29 10:12:19 Using internal plugin for vagrant-cloud
2019/07/29 10:12:19 Using internal plugin for vsphere
2019/07/29 10:12:19 Detected home directory from env var: /home/dummy
2019/07/29 10:12:19 Attempting to open config file: /home/dummy/.packerconfig
2019/07/29 10:12:19 [WARN] Config file doesn't exist: /home/dummy/.packerconfig
2019/07/29 10:12:19 Packer config: &{DisableCheckpoint:false DisableCheckpointSignature:false PluginMinPort:10000 PluginMaxPort:25000 Builders:map[alicloud-ecs:/usr/bin/packer-PACKERSPACE-plugin-PACKERSPACE-packer-builder-alicloud-ecs amazon-chroot:/usr/bin/packer-PACKERSPACE-plugin-PACKERSPACE-packer-builder-amazon-chroot amazon-ebs:/usr/bin/packer-PACKERSPACE-plugin-PACKERSPACE-packer-builder-amazon-ebs amazon-ebssurrogate:/usr/bin/packer-PACKERSPACE-plugin-PACKERSPACE-packer-builder-amazon-ebssurrogate amazon-ebsvolume:/usr/bin/packer-PACKERSPACE-plugin-PACKERSPACE-packer-builder-amazon-ebsvolume amazon-instance:/usr/bin/packer-PACKERSPACE-plugin-PACKERSPACE-packer-builder-amazon-instance arm-image:/usr/bin/packer-builder-arm-image azure-arm:/usr/bin/packer-PACKERSPACE-plugin-PACKERSPACE-packer-builder-azure-arm cloudstack:/usr/bin/packer-PACKERSPACE-plugin-PACKERSPACE-packer-builder-cloudstack digitalocean:/usr/bin/packer-PACKERSPACE-plugin-PACKERSPACE-packer-builder-digitalocean docker:/usr/bin/packer-PACKERSPACE-plugin-PACKERSPACE-packer-builder-docker file:/usr/bin/packer-PACKERSPACE-plugin-PACKERSPACE-packer-builder-file googlecompute:/usr/bin/packer-PACKERSPACE-plugin-PACKERSPACE-packer-builder-googlecompute hcloud:/usr/bin/packer-PACKERSPACE-plugin-PACKERSPACE-packer-builder-hcloud hyperone:/usr/bin/packer-PACKERSPACE-plugin-PACKERSPACE-packer-builder-hyperone hyperv-iso:/usr/bin/packer-PACKERSPACE-plugin-PACKERSPACE-packer-builder-hyperv-iso hyperv-vmcx:/usr/bin/packer-PACKERSPACE-plugin-PACKERSPACE-packer-builder-hyperv-vmcx linode:/usr/bin/packer-PACKERSPACE-plugin-PACKERSPACE-packer-builder-linode lxc:/usr/bin/packer-PACKERSPACE-plugin-PACKERSPACE-packer-builder-lxc lxd:/usr/bin/packer-PACKERSPACE-plugin-PACKERSPACE-packer-builder-lxd ncloud:/usr/bin/packer-PACKERSPACE-plugin-PACKERSPACE-packer-builder-ncloud null:/usr/bin/packer-PACKERSPACE-plugin-PACKERSPACE-packer-builder-null oneandone:/usr/bin/packer-PACKERSPACE-plugin-PACKERSPACE-packer-builder-oneandone openstack:/usr/bin/packer-PACKERSPACE-plugin-PACKERSPACE-packer-builder-openstack oracle-classic:/usr/bin/packer-PACKERSPACE-plugin-PACKERSPACE-packer-builder-oracle-classic oracle-oci:/usr/bin/packer-PACKERSPACE-plugin-PACKERSPACE-packer-builder-oracle-oci parallels-iso:/usr/bin/packer-PACKERSPACE-plugin-PACKERSPACE-packer-builder-parallels-iso parallels-pvm:/usr/bin/packer-PACKERSPACE-plugin-PACKERSPACE-packer-builder-parallels-pvm profitbricks:/usr/bin/packer-PACKERSPACE-plugin-PACKERSPACE-packer-builder-profitbricks proxmox:/usr/bin/packer-PACKERSPACE-plugin-PACKERSPACE-packer-builder-proxmox qemu:/usr/bin/packer-PACKERSPACE-plugin-PACKERSPACE-packer-builder-qemu scaleway:/usr/bin/packer-PACKERSPACE-plugin-PACKERSPACE-packer-builder-scaleway tencentcloud-cvm:/usr/bin/packer-PACKERSPACE-plugin-PACKERSPACE-packer-builder-tencentcloud-cvm triton:/usr/bin/packer-PACKERSPACE-plugin-PACKERSPACE-packer-builder-triton vagrant:/usr/bin/packer-PACKERSPACE-plugin-PACKERSPACE-packer-builder-vagrant virtualbox-iso:/usr/bin/packer-PACKERSPACE-plugin-PACKERSPACE-packer-builder-virtualbox-iso virtualbox-ovf:/usr/bin/packer-PACKERSPACE-plugin-PACKERSPACE-packer-builder-virtualbox-ovf vmware-iso:/usr/bin/packer-PACKERSPACE-plugin-PACKERSPACE-packer-builder-vmware-iso vmware-vmx:/usr/bin/packer-PACKERSPACE-plugin-PACKERSPACE-packer-builder-vmware-vmx yandex:/usr/bin/packer-PACKERSPACE-plugin-PACKERSPACE-packer-builder-yandex] PostProcessors:map[alicloud-import:/usr/bin/packer-PACKERSPACE-plugin-PACKERSPACE-packer-post-processor-alicloud-import amazon-import:/usr/bin/packer-PACKERSPACE-plugin-PACKERSPACE-packer-post-processor-amazon-import artifice:/usr/bin/packer-PACKERSPACE-plugin-PACKERSPACE-packer-post-processor-artifice checksum:/usr/bin/packer-PACKERSPACE-plugin-PACKERSPACE-packer-post-processor-checksum compress:/usr/bin/packer-PACKERSPACE-plugin-PACKERSPACE-packer-post-processor-compress digitalocean-import:/usr/bin/packer-PACKERSPACE-plugin-PACKERSPACE-packer-post-processor-digitalocean-import docker-import:/usr/bin/packer-PACKERSPACE-plugin-PACKERSPACE-packer-post-processor-docker-import docker-push:/usr/bin/packer-PACKERSPACE-plugin-PACKERSPACE-packer-post-processor-docker-push docker-save:/usr/bin/packer-PACKERSPACE-plugin-PACKERSPACE-packer-post-processor-docker-save docker-tag:/usr/bin/packer-PACKERSPACE-plugin-PACKERSPACE-packer-post-proces [... cut ...]
2019/07/29 10:12:19 On error: 
2019/07/29 10:12:19 Waiting on builds to complete...
==> Builds finished but no artifacts were created.
2019/07/29 10:12:19 [INFO] (telemetry) Finalizing.
Failed to initialize build 'arm-image': error initializing builder 'arm-image': fork/exec /usr/bin/packer-builder-arm-image: no such file or directory
arm-image output will be in this color.

==> Builds finished but no artifacts were created.
2019/07/29 10:12:19 waiting for all plugin processes to complete...
```

matt-deboer, 2019-07-29:
Note: building the latest from master doesn't encounter this issue, so maybe just a new release is required?

yuval-k, 2019-11-30:
updated master to work with packer 1.4 and published new binaries. closing as resolved.

closed 2019-11-30 by yuval-k