solo-io/packer-plugin-arm-image #32: arm-image out of space

dehuszar, 2019-10-10:
I have been able to successfully build images around raspbian-buster-lite and write them to sdcards.  That said, I have reached a limit just shy of what I want to add to my arm-image.  I have tried adding a gig to the `target_image_size` param in the example.json `builders` block, but it does not appear to have the effect I was expecting.  is there something else I should be doing to give myself a little extra space to provision a few extra resources?  I shouldn't need more an 200MB more (vim and git, plus plugins and miscellany) than I'm already able to build into my current images.

yuval-k, 2019-10-11:
i think that if you'll make the initial image file bigger first it would have the desired effect (combined with `target_image_size`).
i.e. something like this should work
```
dd if=/dev/zero bs=1024 count=204800  >> my-image-file.img
```

dehuszar, 2019-11-08:
Thanks @insasho, that worked for me.

closed 2019-11-08 by dehuszar