solo-io/packer-plugin-arm-image #34: can not download libreelec image

michaelcohenunsw, 2019-11-13:
example iso_url: "https://releases.libreelec.tv/LibreELEC-RPi2.arm-9.0.2.img.gz"
error:

arm-image output will be in this color.

1 error(s) occurred:

* no image mounts provided. Please set the image mounts or image type.

yuval-k, 2019-11-27:
there's not info info here to understand what's going on. can you provide the full config and command line you use?

jangrewe, 2019-11-28:
without seeing your JSON file it's gonna be difficult, but: i had a similar (the same?) issue and had to use packer 1.3.5 - they changed something with their caching in newer versions, and packer-builder-arm-image isn't compatible with those (yet).

michaelcohenunsw, 2019-11-28:
Hi,

I am using Packer 1.3.5.
Go version 1.13.4

This works:
`{
  "builders": [
    {
      "type": "arm-image",
      "iso_url": "https://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2019-09-30/2019-09-26-raspbian-buster-lite.zip",
      "iso_checksum_type": "sha256",
      "iso_checksum": "a50237c2f718bd8d806b96df5b9d2174ce8b789eda1f03434ed2213bbca6c6ff",
      "mount_path": "/mnt/raspbian",
      "qemu_binary": "qemu-arm-static"
    }
  ]
}`

but this does not:

`{
  "builders": [
    {
      "type": "arm-image",
      "iso_url": "http://releases.libreelec.tv/LibreELEC-RPi2.arm-9.2.0.img.gz",
      "iso_checksum_type": "sha256",
      "iso_checksum": "f9600ac23b8e0f99b8c9069c5136930c3802e3ad351a1f415da3dcf9ae542c0f",
      "mount_path": "/mnt/libreelec",
      "qemu_binary": "qemu-arm-static"
    }
  ]
}`

yuval-k, 2019-11-30:
updated the code to work with packer 1.4.
i think you error is because .gz wasnt supported;
added gzip and bzip support to version 0.1.2

yuval-k, 2019-11-30:
closing as this should be resolved in v0.1.2; please re-open if needed!

closed 2019-11-30 by yuval-k

michaelcohenunsw, 2019-12-02:
I can confirm the arm builder with packer 1.4.5 is able to download https://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2019-09-30/2019-09-26-raspbian-buster-lite.zip.

However, I am getting the same error with http://releases.libreelec.tv/LibreELEC-RPi2.arm-9.2.0.img.gz download.

sergerad, 2020-08-17:
I get this error with `v0.1.6` whenever I change the default `iso_url` value. 

Whether the URL is completely bogus, or points to a valid zipped `.img` file, the same error occurs.