solo-io/packer-plugin-arm-image #35: HOWTO: Building in Docker

jangrewe, 2019-11-28:
Not an issue, but i thought i'd leave it here in case somebody has the same requirement...

I've created a Dockerfile that allows me to build (at least) Raspbian images in a container. I haven't published the image anywhere public, as i don't think there's much use for it on typical cloud CI environments, as the container needs to run `privileged`.

You can find it here: https://git.faked.org/jan/docker-packer-arm

yuval-k, 2019-11-30:
Nice! 
would you like to open a PR with this?  maybe add it under the docker folder?
if you prefer, i can add it myself - let me know.
thank you!

yuval-k, 2020-01-25:
this was solved by #51 
closing.
thanks!

closed 2020-01-25 by yuval-k