solo-io/packer-plugin-arm-image #36: arm-image: chroot: failed to run command ‘/bin/sh’: No such file or directory

lklimek, 2019-12-02:
Looks like some changes introduced on Nov 30 or Dec 1, 2019 have broken the build of raspberry pi 3, 64-bit. We get:

```
==> arm-image: Installing qemu-user-static in the chroot
==> arm-image: Provisioning with shell script: /tmp/packer-shell206115617
    arm-image: chroot: failed to run command ‘/bin/sh’: No such file or directory
```

When using commit b1c268e9d33105634c899e9095f0af097e1af432 , the build goes fine.

Relevant part of our packer configuration:

```
"builders": [{
    "type": "arm-image",
    "iso_url" : "http://cdimage.ubuntu.com/ubuntu-server/bionic/daily-preinstalled/current/bionic-preinstalled-server-arm64+raspi3.img.xz",
    "iso_checksum_type":"sha256",
    "iso_checksum_url": "http://cdimage.ubuntu.com/ubuntu-server/bionic/daily-preinstalled/current/SHA256SUMS",
    "image_mounts": ["/boot/firmware","/"],
    "qemu_binary": "qemu-aarch64-static",
    "chroot_mounts": [
        ["proc", "proc", "/proc"],
        ["sysfs", "sysfs", "/sys"],
        ["bind", "/dev", "/dev"],
        ["devpts", "devpts", "/dev/pts"],
        ["binfmt_misc", "binfmt_misc", "/proc/sys/fs/binfmt_misc"],
        ["bind", "/run/resolvconf", "/run/resolvconf"]
    ]
  }],

```

yuval-k, 2019-12-02:
@lklimek this config on latest master works for me (on fedora); can you provide more information? what host OS are you using? and anything else that might help understand this..

3dc1d3, 2020-02-19:
Getting the same issue.  I tried https://github.com/solo-io/packer-builder-arm-image/commit/b1c268e9d33105634c899e9095f0af097e1af432 as well as latest master (1e1f17b) using `samples/raspbian_golang.json` in each commit.

I'm running this in VirtualBox using packer v1.5.4.

rgl, 2020-05-21:
This is also happening here. I though it was related to https://github.com/solo-io/packer-builder-arm-image/issues/58, but maybe it isn't?

yuval-k, 2020-06-01:
i don't think it's related; and don't have enough information to debug / solve this.

jsiebens, 2020-06-28:
I'm having the same issue when using Ubuntu 18.04. After some tests:

packer: 1.6.0
packer-builder-arm-image: 1.6.0
iso: http://cdimage.ubuntu.com/releases/20.04/release/ubuntu-20.04-preinstalled-server-arm64+raspi.img.xz

Host OS: ubuntu 18.04 -> failure with ‘/bin/sh’: No such file or directory
Host OS: ubuntu 19.10 -> successful build
Host OS: ubuntu 20.04 -> successful build

Hope this give some more insights...

shihanpietersz, 2020-06-28:
I have the same error
    arm-image: chroot: failed to run command ‘/bin/sh’: No such file or directory

even tried with the docker build image and get the same error both for ubuntu 20.04  and raspbian x64.

any help

lklimek, 2020-06-29:
Not sure if it helps, but ensure you have binfmt qemu bindings (`qemu-user-binfmt` on 18.04) installed and configured correctly - there should be file `/proc/sys/fs/binfmt_misc/qemu-aarch64`.

jsiebens, 2020-06-29:
After reading this ([why-cant-ubuntu-18-04-chroot-into-a-qemu-environment](https://askubuntu.com/questions/1204407/why-cant-ubuntu-18-04-chroot-into-a-qemu-environment)) and some debugging, I tried to place the qemu in /usr/bin instead of in the root dir in the chroot. ([here](https://github.com/solo-io/packer-builder-arm-image/blob/1d1a80ad0210a49b78193a1eb506a54bf0c1fb31/pkg/builder/step_qemu_user_static.go#L66)). 
That seems to be working on Ubuntu 18.04. Again, I don't have any problems on 19.10 and 20.04 which are using a newer version of qemu.

As I'm not that familiar with chroot, binfmt, ... I'm not sure if it is the correct solution.
Just sharing my findings ;-)

yuval-k, 2021-11-07:
If you don't need any special arguments to qemu (if you dont know that you need, you probably dont)
Try running this command:
```
/usr/sbin/update-binfmts --enable qemu-aarch64
```
Before running the plugin.

Also note that this bug was fixed in the latest version - 0.2.5

yuval-k, 2021-11-07:
Marking as solved; please re-open if still persists.

closed 2021-11-07 by yuval-k