solo-io/packer-plugin-arm-image #40: error when using .zip download url

michaelcohenunsw, 2019-12-18:
packer version: v1.4.5
arm-image version: v0.1.3
go version go1.13.4 linux/amd64

json config:
`{
  "builders": [
    {
      "type": "arm-image",
      "iso_url": "http://releases.libreelec.tv/LibreELEC-RPi2.arm-9.2.0.img.gz",
      "iso_checksum_type": "sha256",
      "iso_checksum": "f9600ac23b8e0f99b8c9069c5136930c3802e3ad351a1f415da3dcf9ae542c0f",
      "mount_path": "/mnt/libreelec",
      "qemu_binary": "qemu-arm-static"
    }
  ],
  "provisioners": []
}`

`packer build packer-libreelec-usbmedia.json 
arm-image output will be in this color.

1 error(s) occurred:

* no image mounts provided. Please set the image mounts or image type.`

yuval-k, 2019-12-23:
when specifying an image this code will try to detect the image mounts, but if it can't they need to be provided manually.
see here:
https://github.com/solo-io/packer-builder-arm-image/blob/38322f8944181df5c7e943c66a042739caefaa99/pkg/builder/builder.go#L67
and an example here 
https://github.com/solo-io/packer-builder-arm-image/blob/38322f8944181df5c7e943c66a042739caefaa99/pkg/builder/builder.go#L27

yuval-k, 2020-01-25:
closing as answered; feel free to repoen

closed 2020-01-25 by yuval-k