solo-io/packer-plugin-arm-image #46: loopback issue - withing priviliged docker container

cherusk, 2020-01-05:
I'm running packer within priviliged (with --priviliged flag) docker container, which works well for all builders, except for this one:
Getting:

`==> arm-image: https://downloads.raspberrypi.org/raspbian_lite_latest?checksum=sha256%3Aa50237c2f718bd8d806b96df5b9d2174ce8b789eda1f03434ed2213bbca6c6ff => /iso_cache/c16bcbd2827ae3db61afd0321a459237ebebbd75.iso
==> arm-image: Copying source image.
==> arm-image: Image is a zip file.
==> arm-image: Unzipping 2019-09-26-raspbian-buster-lite.img
    arm-image: mappping output-arm-image/image
**==> arm-image: kpartx -s -a -v output-arm-image/image
==> arm-image: error kaprts -l exit status 1: mount: could not find any device /dev/loop#Bad address
==> arm-image: can't set up loop
==> arm-image:**
Build 'arm-image' errored: step canceled or halted

==> Some builds didn't complete successfully and had errors:
--> arm-image: step canceled or halted
`

cherusk, 2020-01-05:
Checked manually in container, with losetup, all worked normally.

cherusk, 2020-01-05:
Solved, was down to missing builder variable: "mount_path": "/mnt"

closed 2020-01-05 by cherusk

yuval-k, 2020-01-28:
i notice this too, just restarting the command seems to solve it for me

silvergasp, 2020-10-16:
I just came across this issue as well, should this not be updated in the README.md