solo-io/packer-plugin-arm-image #48: Is it possible to use this plugin without sudo

joshuaspence, 2020-01-09:
I was looking into whether it would be possible to use this plugin without `sudo packer build`. My biggest problem with the current approach is that the output files and `packer_cache` directories end up being owned by root.

I was looking into whether it would be possible to prefix commands within the plugin with `sudo` so that `sudo` was used only when necessary but this doesn't work for `os.OpenFile` in `pkg/builder/step_register_bin_fmt.go`. Maybe there is a better way though?

yuval-k, 2020-01-14:
yea write to bin_fmt requires root; if there a shell command we can use to do this? this way we can prefix it with sudo.
also, how does this work for you - does sudo not need a password?

joshuaspence, 2020-01-14:
Actually I've started looking into a new approach, using `systemd-nspawn` instead of `chroot`. This would not only remove the need for `sudo` with the `chroot` command but also remove a lot of `mount` commands (as `systemd-nspawn` can handle this as well). I'm not sure if you'd be open to using `systemd-nspawn` so I might fork this plugin instead.

Regarding `step_register_bin_fmt`, does this need to be done every time Packer runs? On my system, the `qemu-user-static` package has a post-install script that sets up `binfmt`:

```
case "$DPKG_MAINTSCRIPT_NAME:$1" in

    postinst:configure)
	for fmt in $fmts ; do

	    eval "case $fmt in $omit) magic= ;; *) magic=\"\$${fmt}_magic\" mask=\"\$${fmt}_mask\" ;; esac"
	    if [ -n "$magic" ]; then
		update-binfmts --package qemu-user-static --install qemu-$fmt /usr/bin/qemu-$fmt-static \
			--magic "$magic" --mask "$mask" --offset 0 --credential yes --fix-binary yes
	    else
		remove_binfmt $fmt
	    fi
	done
	;;

    prerm:remove)
	for fmt in $fmts ; do
	    remove_binfmt $fmt
	done
	;;

esac
```

yuval-k, 2020-01-14:
i'm open to add this as an option (even one that's enabled by default, and assuming it works for everyone deprecate the old way). as for registration, it doesn't need to be done every run, i mainly do that for simplicity for the user; we can add a test if the required value is there, and only write it if its not there

joshuaspence, 2020-01-14:
Thanks for the feedback. I might work on this if I have some spare time :)