solo-io/packer-plugin-arm-image #56: References to go-1.13 seem to break go build in Vagrant

jzr1991, 2020-03-19:
https://github.com/solo-io/packer-builder-arm-image/blob/aafbfd7244266ffc49cb9e2f98dc97a1266b415b/provision-build-env.sh#L50

References to go-1.13 in this file seem to break the Vagrant up build, as it currently installs go-1.14, at least on my machine. Not sure whether you want to lift the version up to 1.14 in this script or create a non-dependent way of doing it.

jzr1991, 2020-03-19:
Or tie the apt-get install down to 1.13.

gtarnaras, 2020-05-23:
@jzr1991 , @yuval-k i have slightly updated the way we're installing Go so that we have control on its version. I included a couple of updates like some cleanup steps and packer version upgrade since the deployment was complaining for an incompatible API version. Please let me know if any comments.

closed 2020-05-26 by yuval-k