solo-io/packer-plugin-arm-image #59: Use local image `file:///` with `docker run`

ChristianKniep, 2020-04-02:
I would like to build an image in multiple steps to not redo all the downloads each time.
I thought I can use `file://` as an `iso_url` and map in the `output-arm-image` directory.

**Question**: How can I reuse an image that I build in a previous run? 

Awesome tool btw; love it!

```
"builders": [
    {
      "type": "arm-image",
      "iso_url": "file:///prom_base.img",
      "last_partition_extra_size": 1048576000,
      "iso_checksum_type": "md5",
      "iso_checksum": "6415daf45a89e98339180bd59de0ec12",
      "image_type": "raspberrypi",
      "image_mounts": ["/output-arm-image", "/"]

    }
  ]
```
When kicking this off it looks like this.
```
docker run \                                                                                                                                    git:(master|✚1…
  --rm --privileged -v ${HOME}/.ssh/:/root/.ssh/ \
  -v ${PWD}:/build:ro \
  -v ${PWD}/packer_cache:/build/packer_cache \
  -v ${PWD}/output-arm-image:/build/output-arm-image  packer-builder-arm build samples/raspbian_prom_setup.json
running /bin/packer
arm-image: output will be in this color.

Warnings for build 'arm-image':

* last_partition_extra_size is deprecated, use target_image_size to grow your image

==> arm-image: Retrieving Image
==> arm-image: Trying file:///prom_base.img
==> arm-image: Trying file:///prom_base.img?checksum=md5%3A6415daf45a89e98339180bd59de0ec12
==> arm-image: Download failed source path error: stat /prom_base.img: no such file or directory
==> arm-image: error downloading Image: [source path error: stat /prom_base.img: no such file or directory]
Build 'arm-image' errored: error downloading Image: [source path error: stat /prom_base.img: no such file or directory]

==> Some builds didn't complete successfully and had errors:
--> arm-image: error downloading Image: [source path error: stat /prom_base.img: no such file or directory]

==> Builds finished but no artifacts were created.
```

yuval-k, 2020-04-02:
images should be cached, even if you use url - do you find this is not the case? are you using the latest version?

ChristianKniep, 2020-04-02:
I want to use the outcome of a previous build to be used in a subsequent one.
The first downloads an image and sets up some basic things and might take a while and the second run will just change some config which should be rather fast.

ChristianKniep, 2020-04-03:
Figured it out... 
I renamed the `image` to `prom_base.img` and ran this.

```
{
  "variables": {},
  "builders": [
    {
      "type": "arm-image",
      "iso_url": "file:////build/output-arm-image/prom_base.img",
      "last_partition_extra_size": 1048576000,
      "iso_checksum_type": "md5",
      "iso_checksum": "6415daf45a89e98339180bd59de0ec12",
      "image_type": "raspberrypi",
      "image_mounts": ["/boot", "/"]

    }
  ],
  "provisioners": [
    {
        "type": "shell",
        "inline": [
            "mkdir -p /opt/build/samples"
        ]
    },{
        "type": "file",
        "source": "/build/samples/",
        "destination": "/opt/build/samples"
    },{
        "type": "shell",
        "inline": [
            "bash -x /opt/build/samples/scripts/prometheus_setup.sh",
            "bash -x /opt/build/samples/scripts/grafana_setup.sh",
            "bash -x /opt/build/samples/scripts/loki_setup.sh"
        ]
    },{
      "type": "shell",
      "inline": [
          "rm -rf /opt/build"
      ]
  }
  ]
}
```
Ran it like that...
```
docker run --rm --privileged
  -v ${PWD}:/build:ro -v ${PWD}/packer_cache:/build/packer_cache \
  -v ${PWD}/output-arm-image:/build/output-arm-image  packer-builder-arm build samples/raspbian_prom_setup.json
```

closed 2020-04-03 by ChristianKniep