solo-io/packer-plugin-arm-image #61: arm-image: read |0: file already closed

micchickenburger, 2020-04-03:
Occasionally arm image builds fail nondeterministically with this error message:

```shell
==> default: Running provisioner: build (shell)...
    default: Running: /var/folders/kd/0qsg7379351cw6j40x_pg_740000gn/T/vagrant-shell20200403-83669-1292gar.sh
    default: '/vagrant/packer-builder-arm-image' -> '/home/vagrant/.packer.d/plugins/packer-builder-arm-image'
    default: Attempting to build image
    default: Debug mode enabled. Builds will not be parallelized.
    default: arm-image: output will be in this color.
    default: ==> arm-image: Retrieving Image
    default: ==> arm-image: Trying http://cdimage.ubuntu.com/releases/18.04.4/release/ubuntu-18.04.4-preinstalled-server-arm64+raspi3.img.xz
    default: ==> arm-image: Trying http://cdimage.ubuntu.com/releases/18.04.4/release/ubuntu-18.04.4-preinstalled-server-arm64+raspi3.img.xz?checksum=sha256%3Af270d4a11fcef7f85ea77bc0642d1c6db2666ae734e9dcc4cb875a31c9f0dc57
    default: ==> arm-image: http://cdimage.ubuntu.com/releases/18.04.4/release/ubuntu-18.04.4-preinstalled-server-arm64+raspi3.img.xz?checksum=sha256%3Af270d4a11fcef7f85ea77bc0642d1c6db2666ae734e9dcc4cb875a31c9f0dc57 => /vagrant/packer_cache/4cc54ff9b15315c3bf2ae2264e8b6d072d8b8e5b.iso
    default: ==> arm-image: Copying source image.
    default: ==> arm-image: Image is a xz file.
    default:     arm-image: Speed:   52.96 MB/s
    default:     arm-image: Speed:   43.34 MB/s
    default:     arm-image: Speed:   43.42 MB/s
    default: ==> arm-image: read |0: file already closed
    default: Build 'arm-image' errored: step canceled or halted
    default: ==> Some builds didn't complete successfully and had errors:
    default: --> arm-image: step canceled or halted
    default: ==> Builds finished but no artifacts were created.
The SSH command responded with a non-zero exit status. Vagrant
assumes that this means the command failed. The output for this command
should be in the log above. Please read the output to determine what
went wrong.
```

I just need to re-run the build one or more times until it works.  I'm not sure where to begin troubleshooting.  Any ideas?

yuval-k, 2020-04-03:
the xz go implementation is really slow, so if available we try to use the command line util. 
can you try commenting out these lines:
https://github.com/solo-io/packer-builder-arm-image/blob/master/pkg/image/image.go#L156-L162

to force it to use the go impl, and see if it  helps?
if it does, we can add a flag to control that behavior

rgl, 2020-05-21:
I had to workaround this by manually (well, in a script) downloading the .xy and re-compressing it as a .zip with `xzcat ubuntu.xz | zip >ubuntu.zip`, and now it never fails with .zip.

yuval-k, 2020-06-08:
good to know, this helps me understand where the bug is

will-dee, 2022-01-12:
Apologies for resurrecting an old issue but I've run into this when using gzip compressed images too. We see the image decomoressing _nearly_ all the way and then the `read |0: file already closed` message and a failed packer run. I think the issue is around this fragment in the `xzFastLane()` method in pkg/image/image.go:

```
go func() {
    xzcat.Wait()
}()
```

The [docs](https://pkg.go.dev/os/exec#Cmd.StdoutPipe) for Cmd.StdoutPipe (which is called a few lines earlier) say:

> Wait will close the pipe after seeing the command exit, so most callers need not close the pipe themselves. It is thus incorrect to call Wait before all reads from the pipe have completed.

The error seems to be caused by this rather than the file handle for the source (non-zip) archive or the destination file being closed prematurely. I have a patch which moves the calling of of `xzcat.Wait()` to after the readers have finished and it appears to fix this issue (at least in the case of gzip compressed images although as far as I can tell the code path is the same for xz images too). I can prepare a pull request for this patch if that is something that you would be interested in reviewing as a possible fix for this issue?

spencergilbert, 2022-04-18:
I'm also running into this and would be happy to test a patch or a PR.

yuval-k, 2022-04-18:
@will-dee a patch would be appriciated

will-dee, 2022-04-21:
I've submitted a [pull request](https://github.com/solo-io/packer-plugin-arm-image/pull/137) containing my fix. In order to test it, I found the bug was most reproducible with a gzip compressed base image. Building a docker image from the current head of master and running this testcase fails fairly consistently with the `read |0: file already closed` error whereas building a docker image with the modified code builds successfully:

```
# Download Latest Raspbian lite image
curl -o 2022-04-04-raspios-bullseye-armhf.img.xz https://downloads.raspberrypi.org/raspios_armhf/images/raspios_armhf-2022-04-07/2022-04-04-raspios-bullseye-armhf.img.xz

# Un-xz the image, gzip it, and checksum it
xz -d -c 2022-04-04-raspios-bullseye-armhf.img.xz | gzip -n > 2022-04-04-raspios-bullseye-armhf.img.gz
shasum -a 256 2022-04-04-raspios-bullseye-armhf.img.gz
  48475ecf26989b30fb2f97a6ae57e3ac5ef972226857556ccd6033d406e677dc

# Docker build and run commands
docker build -t packer-plugin-arm-image:local .
docker run \
  --rm \
  --privileged \
  -v /dev:/dev \
  -v ${PWD}:/build:ro \
  -v ${PWD}/packer_cache:/build/packer_cache \
  -v ${PWD}/output-arm-image:/build/output-arm-image \
  -e PACKER_CACHE_DIR=/build/packer_cache \
  packer-plugin-arm-image:local build samples/raspbian_gz_golang.json
```

The contents of `samples/raspbian_gz_golang.json` is as follows:

```
{
  "variables": {
  },
  "builders": [
    {
      "type": "arm-image",
      "iso_url": "2022-04-04-raspios-bullseye-armhf.img.gz",
      "iso_checksum": "sha256:48475ecf26989b30fb2f97a6ae57e3ac5ef972226857556ccd6033d406e677dc",
      "target_image_size": 5294967296
    }
  ],
  "provisioners": [
    {
      "type": "shell",
      "inline": [
	      "apt-get update",
	      "apt-get install -y golang"
      ]
    }
  ]
}
```

yuval-k, 2022-04-25:
fixed by #137

closed 2022-04-25 by yuval-k