solo-io/packer-plugin-arm-image #78: Is it possible to pass the name for the output image?

christoph-kluge, 2020-06-16:
Thank you for this amazing project! It works out-of-the-box on OS X. I'm using the docker image. Just awesome 👍 🎉 

So as already mentioned in the title I'm searching for a way to define the resulting image name as I would have it differently depending on the variables I'm using. I have checked the `builder.go` but it only references the output-directory of the image and I could'nt find anything related in the packer documentation.

Does somebody has some hints for me? 
(My current alternative is renaming it automatically after the build.)

yuval-k, 2020-06-16:
check `output_filename`

christoph-kluge, 2020-06-16:
Oh snap.. I missed that one. Thank you for the quick support @yuval-k!

closed 2020-06-16 by christoph-kluge