solo-io/packer-plugin-arm-image #83: Error: Plugin not found. Retry build.

crlane, 2020-07-30:
#82 Introduced a breaking bug by changing the path of the plugin installation location. On the very next line, it checks for the location of the packer-builder-arm-image in the old location, causing the script to fail.

```
    default: + PACKERFILE=/vagrant/rpi_ubuntu_arm64_with_salt.json
    default: + sudo mkdir -p /root/.packer.d/plugins
    default: + sudo cp /vagrant/packer-builder-arm-image /root/.packer.d/plugins/
    default: + [[ ! -f /home/vagrant/.packer.d/plugins/packer-builder-arm-image ]]
    default: + echo 'Error: Plugin not found. Retry build.'
    default: Error: Plugin not found. Retry build.
```

crlane, 2020-07-30:
I'm working on a PR, should have one up shortly.

yuval-k, 2020-08-12:
thank you for the fix!

closed 2020-08-12 by yuval-k