solo-io/packer-plugin-arm-image #85: ubuntu/eoan64 no longer available

crlane, 2020-07-30:
The Vagrantfile uses ubuntu/eoan64, which for some reason has been removed from <https://app.vagrantup.com/ubuntu>. This causes the documentation recommended `vagrant up` to fail. 

Additionally, there appears to be a problem with the official ubuntu images for eoan64 and focal64 that prevents vagrant from successfully creating these boxes within the configured time limit. You can read more about it at <https://bugs.launchpad.net/cloud-images/+bug/1874453>. I

While neither of these are problems with this project, the documentation as is is incorrect. If nothing else, hopefully this issue will get picked up by the search index bots and it will be easier for the next person to find the solution. 😄 

As such, the Vagrantfile can be update to the following:

```ruby
# -*- mode: ruby -*-
# vi: set ft=ruby :

Vagrant.configure("2") do |config|
  # for missing eoan64 box
  config.vm.box = "ubuntu/focal64"
  # https://bugs.launchpad.net/cloud-images/+bug/1874453
  config.vm.provider "virtualbox" do |vb|
    vb.customize [ "modifyvm", :id, "--uart1", "0x3F8", "4" ]
    vb.customize [ "modifyvm", :id, "--uartmode1", "file", File.join(Dir.pwd, "ubuntu-focal-20.04-cloudimg-console.log") ]
  end
  config.vm.synced_folder "./", "/vagrant", disabled: false
  config.vm.provision "build-env", type: "shell", :path => "provision-build-env.sh", privileged: false
  config.vm.provision "packer-builder-arm-image", type: "shell", :path => "provision-packer-builder-arm-image.sh", privileged: false, env: {"GIT_CLONE_URL" => ENV["GIT_CLONE_URL"]}
  config.vm.provision "build-image", type: "shell", :path => "provision-build-image.sh", privileged: false, env: {"PACKERFILE" => ENV["PACKERFILE"]}
end
```

yuval-k, 2020-08-12:
i don't have vagrant to to test this, happy to update the Vagrant file if this solves it

fr34k8, 2020-10-21:
@yuval-k @crlane 

just add `bento/ubuntu-18.04` or `bento/ubuntu-20.04`

yuval-k, 2020-10-21:
closing as this was fixed by #90

closed 2020-10-21 by yuval-k