solo-io/packer-plugin-arm-image #86: Changing URL in iso_url fails (v0.1.6)

sergerad, 2020-08-17:
This is the same issue as [this](https://github.com/solo-io/packer-builder-arm-image/issues/34).

I get this error with v0.1.6 whenever I change the default iso_url value.

Whether the URL is completely bogus, or points to a valid zipped .img file, the same error occurs.

Im running on Docker e.g.
```
docker run --rm \
--privileged \
-v ${PWD}:/build:ro \
-v ${PWD}/packer_cache:/build/packer_cache \
-v ${PWD}/output-arm-image:/build/output-arm-image \
docker.pkg.github.com/solo-io/packer-builder-arm-image/packer-builder-arm:${VERSION} build image.json
```

sergerad, 2020-08-18:
Manually adding image_mounts to the JSON fixes the issue:
```
"image_mounts": ["/boot", "/"]
```

However, I then get an odd error considering the zip contains a single `.img` file:
```
==> arm-image: Copying source image.
==> arm-image: Image is a zip file.
==> arm-image: support for only zip files with one file.
Build 'arm-image' errored: step canceled or halted
```

yuval-k, 2020-08-19:
there is auto detection for some urls if the contain certain patterns.
i.e. if your url contains the work "raspberry" then we guess the correct `image_mounts`.

closing as answered

closed 2020-08-19 by yuval-k

whiteadam, 2022-02-25:
I'd like to add that "raspbian" is also properly recognized locally.
eg: `"iso_url": "/build/raspbian-pre-ansible.img"` will detect the correct mount point where `"iso_url": "/build/pre-ansible.img"` will not