solo-io/packer-plugin-arm-image #87: Incompatible API version with plugin

cortlandx, 2020-09-30:
I tried running `packer build samples/raspian_golang.json` after installing the packer-builder-arm-image. I  am getting the following error `Failed to initialize build 'arm-image': error initializing builder 'arm-image': Incompatible API version with plugin.  Plugin version: 5, Ours: 4`. Any assistance on troubleshooting this problem is appreciated.  Thanks.

yuval-k, 2020-10-02:
i think you need to upgrade packer to 1.5.x.
you can also use the docker image as it bundles the correct packer with it.
closing as answered; feel free to re-open if needed

closed 2020-10-02 by yuval-k

cortlandx, 2020-10-05:
Thank you.  That was it.  I didnt notice that when I attempted to added the hashicorp repos that it actually didnt add the repo to my source list.  Once i got it added the issue went away.  Thanks.