solo-io/packer-plugin-arm-image #88: Systemd not running --> Builds but fails at Ansible Provisioner

cortlandx, 2020-10-09:
I am trying to build and provision kalipi on a new Pi4 (64-bit).  My kalipi.json file associated .yml, hosts, and sources.list file are inserted below.  

I am getting the following error (at the ansible provisioner step) after running the `sudo packer build kalipi.json` command: 
``` kalipi: PLAYBOOK: kalipi.yml ***********************************************************
    kalipi: Positional arguments: /tmp/packer-provisioner-ansible-local/5f80990d-fd11-c94d-f631-db6d07a74beb/kalipi.yml
    kalipi: verbosity: 4
    kalipi: connection: local
    kalipi: timeout: 10
    kalipi: become_method: sudo
    kalipi: tags: ('all',)
    kalipi: inventory: ('/tmp/packer-provisioner-ansible-local/5f80990d-fd11-c94d-f631-db6d07a74beb/Hosts',)
    kalipi: extra_vars: ('packer_build_name=kalipi packer_builder_type=arm-image packer_http_addr=%!s(<nil>) -o IdentitiesOnly=yes', 'ansible_python_interpreter=/usr/bin/python3')
    kalipi: forks: 5
    kalipi: 3 plays in /tmp/packer-provisioner-ansible-local/5f80990d-fd11-c94d-f631-db6d07a74beb/kalipi.yml
    kalipi:
    kalipi: PLAY [Play-#1 for building KaliPi4] ********************************************
    kalipi:
    kalipi: TASK [Gathering Facts] *********************************************************
    kalipi: task path: /tmp/packer-provisioner-ansible-local/5f80990d-fd11-c94d-f631-db6d07a74beb/kalipi.yml:2
    kalipi: <127.0.0.1> ESTABLISH LOCAL CONNECTION FOR USER: root
    kalipi: <127.0.0.1> EXEC /bin/sh -c 'echo ~root && sleep 0'
    kalipi: <127.0.0.1> EXEC /bin/sh -c '( umask 77 && mkdir -p "` echo /root/.ansible/tmp `"&& mkdir "` echo /root/.ansible/tmp/ansible-tmp-1602263718.512681-23635-149944798217200 `" && echo ansible-tmp-1602263718.512681-23635-149944798217200="` echo /root/.ansible/tmp/ansible-tmp-1602263718.512681-23635-149944798217200 `" ) && sleep 0'
    kalipi: Using module file /usr/lib/python3/dist-packages/ansible/modules/system/setup.py
    kalipi: <127.0.0.1> PUT /root/.ansible/tmp/ansible-local-23622jewsqgll/tmpnmdnc66q TO /root/.ansible/tmp/ansible-tmp-1602263718.512681-23635-149944798217200/AnsiballZ_setup.py
    kalipi: <127.0.0.1> EXEC /bin/sh -c 'chmod u+x /root/.ansible/tmp/ansible-tmp-1602263718.512681-23635-149944798217200/ /root/.ansible/tmp/ansible-tmp-1602263718.512681-23635-149944798217200/AnsiballZ_setup.py && sleep 0'
    kalipi: <127.0.0.1> EXEC /bin/sh -c '/usr/bin/python3 /root/.ansible/tmp/ansible-tmp-1602263718.512681-23635-149944798217200/AnsiballZ_setup.py && sleep 0'
    kalipi: <127.0.0.1> EXEC /bin/sh -c 'rm -f -r /root/.ansible/tmp/ansible-tmp-1602263718.512681-23635-149944798217200/ > /dev/null 2>&1 && sleep 0'
    kalipi: ok: [pwned1]
    kalipi: META: ran handlers
    kalipi:
    kalipi: TASK [Update Hostname] *********************************************************
    kalipi: task path: /tmp/packer-provisioner-ansible-local/5f80990d-fd11-c94d-f631-db6d07a74beb/kalipi.yml:7
    kalipi: <127.0.0.1> ESTABLISH LOCAL CONNECTION FOR USER: root
    kalipi: <127.0.0.1> EXEC /bin/sh -c 'echo ~root && sleep 0'
    kalipi: <127.0.0.1> EXEC /bin/sh -c '( umask 77 && mkdir -p "` echo /root/.ansible/tmp `"&& mkdir "` echo /root/.ansible/tmp/ansible-tmp-1602263737.0431828-23783-178020833826439 `" && echo ansible-tmp-1602263737.0431828-23783-178020833826439="` echo /root/.ansible/tmp/ansible-tmp-1602263737.0431828-23783-178020833826439 `" ) && sleep 0'
    kalipi: Using module file /usr/lib/python3/dist-packages/ansible/modules/system/hostname.py
    kalipi: <127.0.0.1> PUT /root/.ansible/tmp/ansible-local-23622jewsqgll/tmp7rxnydzq TO /root/.ansible/tmp/ansible-tmp-1602263737.0431828-23783-178020833826439/AnsiballZ_hostname.py
    kalipi: <127.0.0.1> EXEC /bin/sh -c 'chmod u+x /root/.ansible/tmp/ansible-tmp-1602263737.0431828-23783-178020833826439/ /root/.ansible/tmp/ansible-tmp-1602263737.0431828-23783-178020833826439/AnsiballZ_hostname.py && sleep 0'
    kalipi: <127.0.0.1> EXEC /bin/sh -c '/usr/bin/python3 /root/.ansible/tmp/ansible-tmp-1602263737.0431828-23783-178020833826439/AnsiballZ_hostname.py && sleep 0'
    kalipi: <127.0.0.1> EXEC /bin/sh -c 'rm -f -r /root/.ansible/tmp/ansible-tmp-1602263737.0431828-23783-178020833826439/ > /dev/null 2>&1 && sleep 0'
    kalipi: fatal: [pwned1]: FAILED! => {
    kalipi:     "changed": false,
    kalipi:     "invocation": {
    kalipi:         "module_args": {
    kalipi:             "name": "pwned",
    kalipi:             "use": null
    kalipi:         }
    kalipi:     },
    kalipi:     "msg": "Command failed rc=1, out=, err=\u001b[0;1;31mSystem has not been booted with systemd as init system (PID 1). Can't operate.\u001b[0m\n\u001b[0;1;31mFailed to create bus connection: Host is down\u001b[0m\n"
    kalipi: }
    kalipi:
    kalipi: PLAY RECAP *********************************************************************
    kalipi: pwned1                     : ok=1    changed=0    unreachable=0    failed=1    skipped=0    rescued=0    ignored=0
    kalipi:
==> kalipi: fuser -k /tmp/742371882
Build 'kalipi' errored after 7 minutes 15 seconds: Error executing Ansible: Non-zero exit status: 2

```

any thoughts on what my issue maybe is of great assistance.  Of note, it builds and provisions fine with shell commands, but inclusion of the ansible provisioner causes issue.

kalipi.json----------------
```{
    "variables": {},
    "builders": [
        {
            "name": "kalipi",
            "type": "arm-image",
            "iso_url": "https://images.kali.org/arm-images/kali-linux-2020.3a-rpi3-nexmon-64.img.xz",
            "iso_checksum": "56486ff1da0ec5d76b1136863bda079e7d590b9016b5f889b23c443ff4d518de",
[... 145 lines cut ...]
            apt:
                name: "{{ packages.apt.remove }}"
                state: absent
                autoclean: yes
                autoremove: yes
                update_cache: yes
                purge: yes
                force_apt_get: yes
            async: 3600
            poll: 30
            register: apt_sleeper

        -   name: "Reboot expendable post-configuration"
            reboot:
```
host (inventory file)-----
```[kalipi]
pwned1 ansible_host=127.0.0.1 ansible_user=kali ansible_ssh_pass=kali ansible_become_pass=kali ansible_python_interpreter=/usr/bin/python3 new_hostname=pwned
pwned2 ansible_host=127.0.0.1 ansible_user=pwned ansible_ssh_pass=pwned ansible_become_pass=pwned ansible_python_interpreter=/usr/bin/python3
```
sources.list file
```deb http://http.kali.org/kali kali-rolling main contrib non-free
# For source package access, uncomment the following line
# deb-src http://http.kali.org/kali kali-rolling main contrib non-free
```

yuval-k, 2020-10-21:
are you trying to interact with systemd?
the way this plugin works, systemd is not used for 'booting' it; so I'm not sure what we can do here.
would love to get some ideas from someone who knows systemd better

cortlandx, 2020-10-22:
In my ansible file I am using ansible.builtin.hostname function to change the hostname which on kalipi interacts with systemd.  However, yes systemd is not used for "booting" it which is the problem.  However, I did find an easy work around that instead of using ansible.builtin.hostname function I just use Ansible replace function to change the /etc/hostname file similar to how i am using Ansible lineinfile to change /etc/hosts.  

`       -   name: "Change hostname in /etc/hostname"
            replace:
                dest: /etc/hostname
                regexp: "kali"
                replace: "pwned"
`

yuval-k, 2021-07-25:
thanks for figuring this out! closing as answered

closed 2021-07-25 by yuval-k