solo-io/packer-plugin-arm-image #91: `target_image_size`  how to use it  ?

fguiet, 2020-12-01:
Hi,

This is not an issue but I don't understand how to use the property `target_image_size` ?
I want to build a custom image of <https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2020-08-24/2020-08-20-raspios-buster-armhf-lite.zip>

Compressed image size: 432MB

I want to flash this image on a 32gb SD Card....

how to set `target_image_size` property so the last partition fills the 32gb sd card space?

Thank you
Fred

yuval-k, 2020-12-08:
if you are using raspbian, i don't have to use it; as part of the raspbian boot is to make the file system bigger.
This flag is meant to be used if you need a bigger file system as part of building the image.
TL;DR: if you don't have a problem, you don't need to touch this setting

yuval-k, 2021-07-25:
closing as answered, please reopen if needed

closed 2021-07-25 by yuval-k