solo-io/packer-plugin-arm-image #92: Image build failed when I try to install Docker in provisionner part

fguiet, 2020-12-01:
Hi 

I followed this tutorial : <http://bingel.it/raspberry-pi-image-with-hashicorp-packer/>

But it fails .... see below ... any idea?

Info : I use Packer 1.6.5 in the Dockerfile

Here is my json file:

```bash
{
  "variables": {},
  "builders": [{
    "type": "arm-image",
    "iso_url" : "/applications/raspios/2020-08-20-raspios-buster-armhf-lite.zip",
    "iso_checksum": "4522df4a29f9aac4b0166fbfee9f599dab55a997c855702bfe35329c13334668",
    "iso_checksum_type": "sha256"
  }],
  "provisioners": [
    {
      "type": "shell",
      "inline": [
        "touch /boot/ssh",
        "apt-get update",
        "curl -sSL https://get.docker.com | sh &> /dev/null",
        "sudo usermod -aG docker pi"
      ]
    }
  ]
}
```

```bash
root@debian-tools:/applications/packer-builder-arm-image# docker run --rm --privileged -v /applications/packer-builder-arm-image/build:/build -v /applications/packer-builder-arm-image/packer_cache:/build/packer_cache -v /applications/packer-builder-arm-image/build/output-arm-image:/build/output-arm-image -e PACKER_CACHE_DIR=/build/packer_cache -v /applications/raspios:/applications/raspios solo-io-packer-builder-arm build /applications/raspios/raspios2.json
running /bin/packer
arm-image: output will be in this color.

==> arm-image: Retrieving Image
==> arm-image: Trying /applications/raspios/2020-08-20-raspios-buster-armhf-lite.zip
==> arm-image: Trying /applications/raspios/2020-08-20-raspios-buster-armhf-lite.zip?checksum=sha256%3A4522df4a29f9aac4b0166fbfee9f599dab55a997c855702bfe35329c13334668
==> arm-image: /applications/raspios/2020-08-20-raspios-buster-armhf-lite.zip?checksum=sha256%3A4522df4a29f9aac4b0166fbfee9f599dab55a997c855702bfe35329c13334668 => /build/packer_cache/b97f5390e800f79bb942ce5911840e053b371159.iso
==> arm-image: Copying source image.
==> arm-image: Image is a zip file.
==> arm-image: Unzipping 2020-08-20-raspios-buster-armhf-lite.img
    arm-image: Speed:  177.71 MB/s
    arm-image: Progress: 48.15%
    arm-image: Speed:  141.19 MB/s
    arm-image: Progress: 86.40%
    arm-image: mapping output-arm-image/image
==> arm-image: kpartx -s -a -v output-arm-image/image
==> arm-image: partitions: [/dev/mapper/loop0p1 /dev/mapper/loop0p2]
    arm-image: Mounting: /dev/mapper/loop0p2
    arm-image: Mounting: /dev/mapper/loop0p1
==> arm-image: Mounting additional paths within the chroot...
    arm-image: Mounting: /proc
    arm-image: Mounting: /sys
    arm-image: Mounting: /dev
    arm-image: Mounting: /dev/pts
    arm-image: Mounting: /proc/sys/fs/binfmt_misc
==> arm-image: Installing qemu-user-static in the chroot
==> arm-image: Provisioning with shell script: /tmp/packer-shell880311305
    arm-image: Get:1 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
    arm-image: Get:2 http://archive.raspberrypi.org/debian buster InRelease [32.6 kB]
    arm-image: Get:3 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13.0 MB]
    arm-image: Get:4 http://archive.raspberrypi.org/debian buster/main armhf Packages [336 kB]
    arm-image: Get:5 http://raspbian.raspberrypi.org/raspbian buster/contrib armhf Packages [58.7 kB]
    arm-image: Fetched 13.4 MB in 22s (616 kB/s)
    arm-image: Reading package lists...
==> arm-image: sudo: unable to resolve host 02df390c1497: Name or service not known
==> arm-image: usermod: group 'docker' does not exist
    arm-image: # Executing docker install script, commit: 26ff363bcf3b3f5a00498ac43694bf1c7d9ce16c
==> arm-image: + sh -c apt-get update -qq >/dev/null
==> arm-image: + sh -c DEBIAN_FRONTEND=noninteractive apt-get install -y -qq apt-transport-https ca-certificates curl >/dev/null
==> arm-image: + sh -c curl -fsSL "https://download.docker.com/linux/raspbian/gpg" | apt-key add -qq - >/dev/null
==> arm-image: Warning: apt-key output should not be parsed (stdout is not a terminal)
==> arm-image: + sh -c echo "deb [arch=armhf] https://download.docker.com/linux/raspbian buster stable" > /etc/apt/sources.list.d/docker.list
==> arm-image: + sh -c apt-get update -qq >/dev/null
==> arm-image: + [ -n  ]
==> arm-image: + sh -c apt-get install -y -qq --no-install-recommends docker-ce >/dev/null
==> arm-image: debconf: unable to initialize frontend: Dialog
==> arm-image: debconf: (TERM is not set, so the dialog frontend is not usable.)
==> arm-image: debconf: falling back to frontend: Readline
==> arm-image: debconf: unable to initialize frontend: Readline
==> arm-image: debconf: (This frontend requires a controlling tty.)
==> arm-image: debconf: falling back to frontend: Teletype
==> arm-image: dpkg-preconfigure: unable to re-open stdin:
==> arm-image: E: Sub-process /usr/bin/dpkg returned an error code (1)
==> arm-image: fuser -k /tmp/272101727
Build 'arm-image' errored after 2 minutes 51 seconds: Script exited with non-zero exit status: 6.Allowed exit codes are: [0]

==> Wait completed after 2 minutes 51 seconds

==> Some builds didn't complete successfully and had errors:
--> arm-image: Script exited with non-zero exit status: 6.Allowed exit codes are: [0]

==> Builds finished but no artifacts were created.
```

yuval-k, 2020-12-08:
i think from the output the package is trying to ask you a question, and as there is no tty it fails to do so. Is there way to install it in an uninteractive way?

yuval-k, 2021-07-25:
Hi, closing this as inactive, please re-open if needed

closed 2021-07-25 by yuval-k

jseparovic, 2022-04-29:
I just had this same issue. When I removed all the output suppression from the get docker script, it showed me that the device was full. I hadn't set a target_image_size. I updated the builder config to `"target_image_size": 4294967296` and docker installed no worries. Just thought I'd leave this here incase someone else stumbles across it with the same issue.