solo-io/packer-plugin-arm-image #94: packer unable to find local file

gamename, 2021-02-05:
Hi,

I'm getting this error when trying to build an image with docker.  The file it cannot find ("/Users/tennis.smith/.ssh/id_rsa.pub") absolutely exists.  It appears the utility is looking for the file inside the container instead of externally.

Error:
```
running /bin/packer
Error: Failed to prepare build: "arm-image"

1 error(s) occurred:

* Bad source '/Users/tennis.smith/.ssh/id_rsa.pub': stat
/Users/tennis.smith/.ssh/id_rsa.pub: no such file or directory
```

Here is the command:
```
docker run \
  --rm \
  --privileged \
  -v ${PWD}:/build:ro \
  -v ${PWD}/packer_cache:/build/packer_cache \
  -v ${PWD}/output-arm-image:/build/output-arm-image \
  docker.pkg.github.com/solo-io/packer-builder-arm-image/packer-builder-arm build \
  -var wifi_name=foo -var wifi_password=bar -var home=/Users/tennis.smith \
  ./pi-set-wifi-and-ssh.json
```

Here is the input JSON file:
```json
{
  "variables": {
    "wifi_name": "",
    "wifi_password": "",
    "home": ""
  },
  "builders": [{
    "type": "arm-image",
    "iso_url": "https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2020-08-24/2020-08-20-raspios-buster-arm64-lite.zip",
    "iso_checksum": "sha256:0639c516aa032df314b176bda97169bdc8564e7bc5afd4356caafbc3f6d090ed",
    "last_partition_extra_size": 1073741824
  }],
  "provisioners": [
    {
      "type": "shell",
      "inline": ["touch /boot/ssh"]
    },
    {
      "type": "shell",
      "inline": [
        "wpa_passphrase \"{{user `wifi_name`}}\" \"{{user `wifi_password`}}\" | sed -e 's/#.*$//' -e '/^$/d' >> /etc/wpa_supplicant/wpa_supplicant.conf"
      ]
    },
    {
      "type": "file",
      "source": "{{user `home`}}/.ssh/id_rsa.pub",
      "destination": "/home/pi/.ssh/authorized_keys"
    },
    {
      "type": "shell",
      "inline": [
        "sed '/PasswordAuthentication/d' -i /etc/ssh/sshd_config",
        "echo  >> /etc/ssh/sshd_config",
        "echo 'PasswordAuthentication no' >> /etc/ssh/sshd_config"
      ]
    }
  ]
}
```

Here is my docker version:
```log
$ docker --version
Docker version 19.03.13, build 4484c46d9d
```

yuval-k, 2021-02-09:
because you are running this using docker, the path needs to be accessible form the docker vm in the mac;
i'm not sure how mounting the home dir in mac works in docker as it is a bit magical...
but a workaround can be mounting the file to a known location in docker (i.e. add `-v /Users/tennis.smith/.ssh/id_rsa.pub:/tmp/id_rsa.pub`) and use this path (`/tmp/id_rsa.pub` in our example) in your packer config.

note that i did not test this as i'm a linux user.

gamename, 2021-02-21:
@yuval-k 

I tried your suggestions and they didn't work. :( I gave up and closed the issue.

But later I thought I would try one more time, but with the vagrant approach this time.

Again the utility couldn't find my `id_rsa.pub` file.  So, I added the pub file to a special dir inside the vagrant directory.  That was accepted by packer, but the build still failed on the `Upload` section (see below).  There is also an error early on in the build as well. I'll mark both with `--->` eye catchers. 

```log
vagrant provision --provision-with build-image
==> default: Running provisioner: build-image (shell)...
    default: Running: /var/folders/h1/f3ncj_2s7bqfmm3vpxbbm1h00000gn/T/vagrant-shell20210221-66971-1n91nk2.sh
    default: + set -e
    default: + export PACKERFILE=samples/pi-secure-wifi-ssh.json
    default: + PACKERFILE=samples/pi-secure-wifi-ssh.json
    default: + PLUGIN_DIR=/root/.packer.d/plugins
    default: + sudo mkdir -p /root/.packer.d/plugins
    default: + sudo cp /vagrant/packer-builder-arm-image /root/.packer.d/plugins/
    default: + sudo '[[' '!' -f /root/.packer.d/plugins/packer-builder-arm-image ']]'
--->    default: sudo: [[: command not found
    default: Attempting to build image
    default: + echo 'Attempting to build image'
    default: ++ mktemp
    default: + PACKER_LOG=/tmp/tmp.0IG2bq2nJx
    default: + [[ -f /vagrant/samples/pi-secure-wifi-ssh.json ]]
    default: + sudo packer build /vagrant/samples/pi-secure-wifi-ssh.json
    default: + tee /tmp/tmp.0IG2bq2nJx
    default: arm-image: output will be in this color.
    default: Warnings for build 'arm-image':
    default:
    default: * last_partition_extra_size is deprecated, use target_image_size to grow your image
    default:
    default: ==> arm-image: Retrieving Image
    default: ==> arm-image: Trying https://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2020-02-14/2020-02-13-raspbian-buster-lite.zip
    default: ==> arm-image: Trying https://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2020-02-14/2020-02-13-raspbian-buster-lite.zip?checksum=sha256%3A12ae6e17bf95b6ba83beca61e7394e7411b45eba7e6a520f434b0748ea7370e8
    default: ==> arm-image: https://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2020-02-14/2020-02-13-raspbian-buster-lite.zip?checksum=sha256%3A12ae6e17bf95b6ba83beca61e7394e7411b45eba7e6a520f434b0748ea7370e8 => /home/vagrant/packer_cache/936f342e729d0d65e1139ae318d05be6a1e1c5a4.iso
    default: ==> arm-image: Copying source image.
    default: ==> arm-image: Image is a zip file.
    default: ==> arm-image: Unzipping 2020-02-13-raspbian-buster-lite.img
    default:     arm-image: Speed:  140.22 MB/s
    default:     arm-image: Progress: 37.91%
    default:     arm-image: Speed:   84.33 MB/s
    default:     arm-image: Progress: 60.70%
    default:     arm-image: Speed:  109.16 MB/s
    default:     arm-image: Progress: 90.22%
    default: ==> arm-image: Growing partition with 1024 M (1073741824 bytes)
    default:     arm-image: mapping output-arm-image/image
    default: ==> arm-image: kpartx -s -a -v output-arm-image/image
    default: ==> arm-image: partitions: [/dev/mapper/loop3p1 /dev/mapper/loop3p2]
    default: ==> arm-image: partitions: [/dev/mapper/loop3p1 /dev/mapper/loop3p2]
    default:     arm-image: Mounting: /dev/mapper/loop3p2
    default:     arm-image: Mounting: /dev/mapper/loop3p1
    default: ==> arm-image: Mounting additional paths within the chroot...
    default:     arm-image: Mounting: /proc
    default:     arm-image: Mounting: /sys
    default:     arm-image: Mounting: /dev
    default:     arm-image: Mounting: /dev/pts
    default:     arm-image: Mounting: /proc/sys/fs/binfmt_misc
    default: ==> arm-image: Installing qemu-user-static in the chroot
    default: ==> arm-image: Provisioning with shell script: /tmp/packer-shell174857660
    default: ==> arm-image: Provisioning with shell script: /tmp/packer-shell736009096
    default: ==> arm-image: Uploading /vagrant/ssh/.ssh/id_rsa.pub => /home/pi/.ssh/authorized_keys
--->    default: ==> arm-image: Upload failed: exit status 1
id_rsa.pub 748 B / 748 B  100.00% 0s
id_rsa.pub 748 B / 748 B  100.00% 0s
    default: ==> arm-image: fuser -k /tmp/065409010
    default: Build 'arm-image' errored: exit status 1
    default:
    default: ==> Some builds didn't complete successfully and had errors:
    default: --> arm-image: exit status 1
    default:
    default: ==> Builds finished but no artifacts were created.
    default: ++ grep -Po '(?<=Build '\'').*(?='\'' finished.)' /tmp/tmp.0IG2bq2nJx
    default: + BUILD_NAME=
The SSH command responded with a non-zero exit status. Vagrant
assumes that this means the command failed. The output for this command
should be in the log above. Please read the output to determine what
went wrong.
```