solo-io/packer-plugin-arm-image #96: Incompatibility with Packer 1.7

sprehn, 2021-03-08:
Hi,
just installed a fresh ubuntu server 20.10
plus, packer following official tutorial at https://learn.hashicorp.com/tutorials/packer/getting-started-install
when I run a build it fails with following error:

```
Error: Failed to initialize build "arm-image"

error initializing builder 'arm-image': The protocol of this plugin (protocol
version 4 and lower) was deprecated, please use a newer version of this
plugin.Or use an older version of Packer (pre 1.7) with this plugin.
```

It looks like there is an incompatibility with the latest official release of packer.

rgl, 2021-03-11:
I would also like to use packer 1.7.

This seems to be addressed by https://github.com/solo-io/packer-builder-arm-image/pull/95.

yuval-k, 2021-03-27:
yes packer 1.7 had breaking changes. there's a PR in progress

yuval-k, 2021-07-08:
this should be resolved now

closed 2021-07-08 by yuval-k