solo-io/packer-plugin-arm-image #97: Can't use Ansible with Ubuntu 20.10

nfx, 2021-03-17:
It seems that process is stuck waiting for SSH server to start in Quemu, but it won't start in 15 minutes and timeout. What might be missing?

```json
{
  "builders": [
    {
      "type": "arm-image",
      "image_type": "raspberrypi",
      "iso_url": "https://cdimage.ubuntu.com/releases/20.10/release/ubuntu-20.10-preinstalled-server-arm64+raspi.img.xz",
      "iso_checksum": "sha256:73a98e83fbb6398f86902c7a83c826536f1afbd0be2d80cb0d7faa94ede33137",
      "image_mounts": ["/boot/firmware","/"],
      "chroot_mounts": [
        ["proc", "proc", "/proc"],
        ["sysfs", "sysfs", "/sys"],
        ["bind", "/dev", "/dev"],
        ["devpts", "devpts", "/dev/pts"],
        ["binfmt_misc", "binfmt_misc", "/proc/sys/fs/binfmt_misc"]
      ]
    }
  ],
  "provisioners": [
    {
      "type": "shell",
      "inline": ["touch /boot/ssh"]
    },
    {
      "type": "ansible",
      "extra_arguments": ["-vvvv"],
      "playbook_file": "./ansible/make-image.yml"
    }
  ]
}
```

yuval-k, 2021-03-27:
i'm not very familiar with ansible. but running images with this tool does not start init.d services like ssh, so that's probalby why it's waiting. Think of this tool a more similar flow to `docker build`. you can run programs and copy files, but system services won't be started automatically.

nfx, 2021-03-27:
it has something to do with ssh started by quemu, as i've figured out later..

yuval-k, 2021-03-29:
sounds like this is resolved. closing

closed 2021-03-29 by yuval-k