solo-io/packer-plugin-arm-image #99: Unknown address type: 0 error with Vagrant

asfernandes, 2021-06-11:
```
==> default: Running provisioner: build-image (shell)...
    default: Running: /tmp/vagrant-shell20210611-410902-pykxw4.sh
    default: + set -e
    default: + export PACKERFILE=samples/raspbian_golang.json
    default: + PACKERFILE=samples/raspbian_golang.json
    default: + PLUGIN_DIR=/root/.packer.d/plugins
    default: + sudo mkdir -p /root/.packer.d/plugins
    default: + sudo cp /vagrant/packer-builder-arm-image /root/.packer.d/plugins/
    default: + sudo '[[' '!' -f /root/.packer.d/plugins/packer-builder-arm-image ']]'
    default: sudo: [[: command not found
    default: Attempting to build image
    default: + echo 'Attempting to build image'
    default: ++ mktemp
    default: + PACKER_LOG=/tmp/tmp.MGsup7Yvuw
    default: + [[ -f /vagrant/samples/raspbian_golang.json ]]
    default: + tee /tmp/tmp.MGsup7Yvuw
    default: + sudo packer build /vagrant/samples/raspbian_golang.json
    default: Failed to initialize build 'arm-image': error initializing builder 'arm-image': Unknown address type: 0
    default: ==> Builds finished but no artifacts were created.
    default: ++ grep -Po '(?<=Build '\'').*(?='\'' finished.)' /tmp/tmp.MGsup7Yvuw
    default: + BUILD_NAME=
The SSH command responded with a non-zero exit status. Vagrant
assumes that this means the command failed. The output for this command
should be in the log above. Please read the output to determine what
went wrong.
```

vincentracine, 2021-06-19:
I started seeing the same thing, MacOS Big Sur building via Docker:

```
Running /bin/packer build -debug -force templates/automify-os.json
2021/06/19 21:25:23 [INFO] Packer version: 1.6.0 [go1.13.12 linux amd64]
2021/06/19 21:25:23 [DEBUG] Discovered plugin: arm-image = /usr/bin/packer-builder-arm-image
2021/06/19 21:25:23 using external builders [arm-image]
2021/06/19 21:25:23 Checking 'PACKER_CONFIG' for a config file path
2021/06/19 21:25:23 'PACKER_CONFIG' not set; checking the default config file path
2021/06/19 21:25:23 Attempting to open config file: /root/.packerconfig
2021/06/19 21:25:23 [WARN] Config file doesn't exist: /root/.packerconfig
2021/06/19 21:25:23 Setting cache directory: /build/packer_cache
Running in background, not using a TTY
2021/06/19 21:25:23 Creating plugin client for path: /usr/bin/packer-builder-arm-image
2021/06/19 21:25:23 Starting plugin: /usr/bin/packer-builder-arm-image []string{"/usr/bin/packer-builder-arm-image"}
2021/06/19 21:25:23 Waiting for RPC address for: /usr/bin/packer-builder-arm-image

error initializing builder 'arm-image': Unknown address type: 0

2021/06/19 21:25:23 packer-builder-arm-image plugin: 2021/06/19 21:25:23 Plugin address: unix /root/packer_tmp/packer-plugin381278970
2021/06/19 21:25:23 Build debug mode: true
2021/06/19 21:25:23 Force build: true
2021/06/19 21:25:23 On error: 
2021/06/19 21:25:23 packer-builder-arm-image plugin: 2021/06/19 21:25:23 Waiting for connection...
2021/06/19 21:25:23 Waiting on builds to complete...
==> Builds finished but no artifacts were created.
2021/06/19 21:25:23 [INFO] (telemetry) Finalizing.
Error: Failed to initialize build "arm-image"

error initializing builder 'arm-image': Unknown address type: 0

2021/06/19 21:25:23 /usr/bin/packer-builder-arm-image: plugin process exited
Debug mode enabled. Builds will not be parallelized.

==> Builds finished but no artifacts were created.
2021/06/19 21:25:24 waiting for all plugin processes to complete...
```

blake, 2021-06-24:
When Packer executes a plugin, the plugin responds with a string that contains the `<version>|<RPC address family>|<RPC address>`.

The version format changed between v4 and v5, where v5 now also includes a minor version (i.e., `5.0`). This comment in https://github.com/hashicorp/packer-plugin-sdk/blob/v0.0.14/plugin/server.go#L37-L54 notes that

When Packer connects to Version 5 is only supported in Packer 1.7.0 and later. which uses a different API version format

This error is raised because Packer is unable to parse the plugin's

blake, 2021-06-24:
I was able to figure out the cause of this error.

When Packer executes a plugin, the plugin responds with a string that contains the supported Packer API version, and the RPC address at which to reach the plugin.

PR #95 recently updated this plugin to use a newer version of the Packer plugin SDK, v0.0.14, which produces a string which is incompatible with Packer 1.5.3.

The old format, and that which is expected by Packer 1.5.3, is `<version>|<RPC address family>|<RPC address>`.
The updated format, which is compatible with Packer 1.7.0, is `<major_ver>|<min_ver>|<RPC address family>|<RPC address>`.

As noted in the comment in https://github.com/hashicorp/packer-plugin-sdk/blob/v0.0.14/plugin/server.go#L37-L54 ...

> Api version 4 did not have the notion of a minor version, so Packer will error with a weird error message.

That weird error, `Unknown address type: 0`, is raised because Packer 1.5.3 is parsing the minor API version as the address family type, due to the format change (https://github.com/hashicorp/packer/blob/v1.5.3/packer/plugin/client.go#L311-L336). The supported address family types are `tcp` and `unix`, not `0`.

The solution is simply to update `provision-build-env.sh` to install Packer 1.7.x or later so that this string is correctly parsed by Packer, and the plugin can be executed. I have a branch which does this at blake/packer-builder-arm-image@e37f20723c80dcb5ba10c577f977aac8576356e9.

I also made a few changes to resolve errors in the provisioning scripts when executing them the [ubuntu/focal64 v20210622.0.0](https://app.vagrantup.com/ubuntu/boxes/focal64/versions/20210622.0.0) Vagrant Box. I'm able to successfully build an ARM image using `vagrant up` after incorporating these changes.

kerkerj, 2021-07-04:
@blake https://github.com/solo-io/packer-builder-arm-image/pull/101 works well and solved the `Unknown address type: 0 error` issue that I faced as well 👍

PR #101 (2021-07-08): Update Vagrantfile to install Packer 1.7.3
Update Vagrantfile to install Packer 1.7.3.

This change also resolves several errors in the bash script which prevented the relevant commands from being executed.

Resolves #99
closed 2021-07-08 by yuval-k