solo-io/packer-plugin-arm-image #105: chroot: failed to run command '/bin/sh': Exec format error

groldo, 2021-09-16:
Hi,

I'm running packer from docker on a virtualized ubuntu:

```bash
docker run --rm --privileged -v ${PWD}:/build:ro -v ${PWD}/packer_cache:/build/packer_cache -v ${PWD}/output-arm-image:/build/output-arm-image quay.io/solo-io/packer-builder-arm-image:v0.1.5 build samples/raspiOS_cloud-init_armhf_64bit.json
```

Everything works as expected with armhf from raspberrypi.org.
Now when I try to do the same with arm64:

```
arm-image: output will be in this color.

==> arm-image: Retrieving Image
==> arm-image: Trying https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2021-05-28/2021-05-07-raspios-buster-arm64-lite.zip
==> arm-image: Trying https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2021-05-28/2021-05-07-raspios-buster-arm64-lite.zip?checksum=sha256%3A868cca691a75e4280c878eb6944d95e9789fa5f4bbce2c84060d4c39d057a042
==> arm-image: https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2021-05-28/2021-05-07-raspios-buster-arm64-lite.zip?checksum=sha256%3A868cca691a75e4280c878eb6944d95e9789fa5f4bbce2c84060d4c39d057a042 => /build/packer_cache/e993b162855163ced2b2ce5f729e4142b5e82b33.iso
==> arm-image: Copying source image.
==> arm-image: Image is a zip file.
==> arm-image: Unzipping 2021-05-07-raspios-buster-arm64-lite.img
    arm-image: Speed:   87.16 MB/s
    arm-image: Progress: 24.00%
    arm-image: Speed:   34.08 MB/s
    arm-image: Progress: 33.38%
    arm-image: Speed:   33.91 MB/s
    arm-image: Progress: 42.71%
    arm-image: Speed:   39.78 MB/s
    arm-image: Progress: 53.68%
    arm-image: Speed:   43.67 MB/s
    arm-image: Progress: 65.69%
    arm-image: Speed:   48.65 MB/s
    arm-image: Progress: 79.08%
==> arm-image: Growing partition to 4096 M (4294967296 bytes)
    arm-image: mapping output-arm-image/image
==> arm-image: kpartx -s -a -v output-arm-image/image
==> arm-image: partitions: [/dev/mapper/loop8p1 /dev/mapper/loop8p2]
==> arm-image: partitions: [/dev/mapper/loop8p1 /dev/mapper/loop8p2]
    arm-image: Mounting: /dev/mapper/loop8p2
    arm-image: Mounting: /dev/mapper/loop8p1
==> arm-image: Mounting additional paths within the chroot...
    arm-image: Mounting: /proc
    arm-image: Mounting: /sys
    arm-image: Mounting: /dev
    arm-image: Mounting: /dev/pts
    arm-image: Mounting: /proc/sys/fs/binfmt_misc
==> arm-image: Installing qemu-user-static in the chroot
==> arm-image: Uploading /build/scripts/sshd_config => /etc/ssh/sshd_config
sshd_config 3.21 KiB / 3.21 KiB  100.00% 0s
==> arm-image: Uploading /build/scripts/locale.gen => /etc/locale.gen
locale.gen 161 B / 161 B  100.00% 0s
==> arm-image: Provisioning with shell script: /tmp/packer-shell460726097
==> arm-image: chroot: failed to run command '/bin/sh': Exec format error
==> arm-image: fuser -k /tmp/543292909
Build 'arm-image' errored: Script exited with non-zero exit status: 126.Allowed exit codes are: [0]

==> Some builds didn't complete successfully and had errors:
--> arm-image: Script exited with non-zero exit status: 126.Allowed exit codes are: [0]

==> Builds finished but no artifacts were created.
```

Am I do something wrong here? This is the json:

```json
{
  "variables": {
  },
  "builders": [
    {
      "type": "arm-image",
      "image_type": "raspberrypi",
      "qemu_binary": "qemu-aarch64-static",
      "iso_url": "https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2021-05-28/2021-05-07-raspios-buster-arm64-lite.zip",
      "iso_checksum": "sha256:868cca691a75e4280c878eb6944d95e9789fa5f4bbce2c84060d4c39d057a042",
      "target_image_size": 4294967296
    }
  ],
  "provisioners": [
    {
      "type": "file",
      "source": "/build/scripts/sshd_config",
      "destination": "/etc/ssh/sshd_config"
    },
    {
      "type": "file",
      "source": "/build/scripts/locale.gen",
      "destination": "/etc/locale.gen"
    },
    {
      "type": "shell",
      "inline": [
        "locale-gen",
        "systemctl enable ssh",
        "apt-get update --allow-releaseinfo-change",
        "apt-get -yy upgrade",
	      "apt-get install -y cloud-init ufw fail2ban vim",
        "rm /etc/cloud/cloud.cfg.d/*"
      ]
    },
    {
      "type": "file",
      "source": "/build/scripts/99-disable-network-config.cfg",
      "destination": "/etc/cloud/cloud.cfg.d/99-disable-network-config.cfg"
    },
    {
      "type": "file",
      "source": "/build/scripts/90_custom.cfg",
      "destination": "/etc/cloud/cloud.cfg.d/90_config.cfg"
    }
  ]
}
```

yuval-k, 2021-09-20:
you may want to set `qemu_binary` to `qemu-aarch64-static` in the [builder setttings](https://github.com/solo-io/packer-builder-arm-image/blob/920cc8d3c01eb3f3a3889ec63441924de963b858/pkg/builder/builder.go#L118).

yuval-k, 2021-09-20:
it might be that `binfmt_misc` is not setup correctly. it is handled here: https://github.com/solo-io/packer-builder-arm-image/blob/6238af4651411e4668218fcfdc4957cd693cd043/pkg/builder/step_register_bin_fmt.go#L22 

Unfortunately I dont have the bandwith to take a deeper look but hopefully this can help

yuval-k, 2021-11-02:
it's most likely losetup didn't detect partitions (i should probably add that check in that step). 
are you able to download raspbian & unzip, and then paste  the output you are seeing from 
`losetup --show -f -P <RPI IMAGE.img>`
?

yuval-k, 2021-11-07:
also please note if any new loop devices were created. you can see using the output of `ls /dev/loop*`

groldo, 2021-11-28:
Reporting back:

```bash
vagrant@hosttwo:~/Documents/Projects$ wget https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2021-05-28/2021-05-07-raspios-buster-arm64-lite.zip
[truncated]
‘2021-05-07-raspios-buster-arm64-lite.zip’ saved

vagrant@hosttwo:~/Documents/Projects$ unzip 2021-05-07-raspios-buster-arm64-lite.zip 
Archive:  2021-05-07-raspios-buster-arm64-lite.zip
  inflating: 2021-05-07-raspios-buster-arm64-lite.img  
vagrant@hosttwo:~/Documents/Projects$ losetup --show -f -P 2021-05-07-raspios-buster-arm64-lite.img 
losetup: 2021-05-07-raspios-buster-arm64-lite.img: failed to set up loop device: Permission denied
vagrant@hosttwo:~/Documents/Projects$ sudo losetup --show -f -P 2021-05-07-raspios-buster-arm64-lite.img 
/dev/loop0
vagrant@hosttwo:~/Documents/Projects$ sudo losetup --show -f -P 2021-05-07-raspios-buster-arm64-lite.img 
/dev/loop1
```

I did it twice as a first didn't notice it actually was returning something.
now `ls -lah /dev/loop*`

```bash
vagrant@hosttwo:~/Documents/Projects$ ls -lah /dev/loop*
brw-rw---- 1 root disk   7,   0 Nov 28 11:11 /dev/loop0
brw-rw---- 1 root disk 259,   0 Nov 28 11:11 /dev/loop0p1
brw-rw---- 1 root disk 259,   1 Nov 28 11:11 /dev/loop0p2
brw-rw---- 1 root disk   7,   1 Nov 28 11:11 /dev/loop1
brw-rw---- 1 root disk 259,   2 Nov 28 11:11 /dev/loop1p1
brw-rw---- 1 root disk 259,   3 Nov 28 11:11 /dev/loop1p2
brw-rw---- 1 root disk   7,   2 Nov 27 17:31 /dev/loop2
brw-rw---- 1 root disk   7,   3 Nov 27 17:31 /dev/loop3
brw-rw---- 1 root disk   7,   4 Nov 27 17:31 /dev/loop4
brw-rw---- 1 root disk   7,   5 Nov 27 17:31 /dev/loop5
brw-rw---- 1 root disk   7,   6 Nov 27 17:31 /dev/loop6
brw-rw---- 1 root disk   7,   7 Nov 27 17:31 /dev/loop7
crw-rw---- 1 root disk  10, 237 Nov 27 17:04 /dev/loop-control
```

Hope this is helpful.

yuval-k, 2021-12-07:
@groldo i think i see the issue, please add `-v /dev:/dev` to your docker run arguments (we updated the readme recently)