solo-io/packer-plugin-arm-image #114: error initializing builder 'arm-image': Unknown builder arm-image

5t33, 2021-10-18:
Simply ran

```
docker build . -t packer-builder-arm
```

then 

```
cd samples
docker run \
  --rm \
  --privileged \
  -v ${PWD}:/build:ro \
  -v ${PWD}/packer_cache:/build/packer_cache \
  -v ${PWD}/output-arm-image:/build/output-arm-image \
  -e PACKER_CACHE_DIR=/build/packer_cache \
  packer-builder-arm build raspbian_golang.json
```

and got 

```
Error: Failed to initialize build "arm-image"

error initializing builder 'arm-image': Unknown builder arm-image
```

5t33, 2021-10-21:
Odd. Mac or Linux?

eaglesemanation, 2021-10-21:
@5t33 Linux, specifically Fedora 34. Could you specify which image version you used?

squadramunter, 2021-11-04:
I have the same problem and I am running this in a standalone mode.

`  },
  "builders": [{
    "type": "arm-image",
    "image_type": "raspberrypi",
    "iso_url": "http://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2020-08-24/2020-08-20-raspios-buster-arm64-lite.zip",
    "iso_checksum_type": "sha256",
    "iso_checksum": "0639c516aa032df314b176bda97169bdc8564e7bc5afd4356caafbc3f6d090ed"
  }],`

Do I something wrong in this config?

I am running this on Ubuntu 20.04.3 LTS with kernel Linux ubuntu 5.11.0-38-generic #42~20.04.1-Ubuntu

eaglesemanation, 2021-11-04:
@squadramunter do you run it containerized, for example inside Docker or maybe Snap? If so, can you try running it with Packer v1.7.0 or higher? To do it without manually installing this plugin you will need to do configuration in HCL instead of JSON and add this: 
```hcl
packer {
  required_plugins {
    arm-image = {
      version = ">= 0.2.5"
      source  = "github.com/solo-io/arm-image"
    }
  }
}
```

squadramunter, 2021-11-04:
I know what I did wrong. I did not copy the file packer-plugin-arm-image after I built it using go to my /usr/local/bin folder. Now it works haha

5t33, 2021-11-22:
Closing because I can't recreate

closed 2021-11-22 by 5t33