solo-io/packer-plugin-arm-image #121: Please add MountPath to build variables

monester, 2021-11-18:
There is possible to expose context variables from builders, https://www.packer.io/docs/templates/hcl_templates/contextual-variables#build-variables.

Can you add MountPath variable to mounted root like it is done in amazon-chroot builder https://www.packer.io/docs/builders/amazon/chroot#build-shared-information-variables

It will make possible to use ansible provisioner instead of running local-shell with full ansible command like it is done in example https://github.com/solo-io/packer-plugin-arm-image/blob/master/samples/raspbian_ansible_chroot.json

And replace with something like this one:
```hcl
build {
  name = "test"

  source "arm-image.raspbian-os" {}

  provisioner "ansible" {
    playbook_file = "/playbook.yml"
    extra_arguments = [
      "-e", "ansible_connection=chroot ansible_ssh_host=${build.MountPath}",
    ]
  }
}
```

yuval-k, 2022-01-25:
fixed by https://github.com/solo-io/packer-plugin-arm-image/pull/125

closed 2022-01-25 by yuval-k

OGKevin, 2022-02-19:
Can we have a release that includes this fix 🙏🏾