solo-io/packer-plugin-arm-image #122: no partitions defined

groldo, 2021-12-07:
When running 

```bash
sudo docker run --rm --privileged -v ${PWD}:/build:ro -v ${PWD}/packer_cache:/build/packer_cache -v ${PWD}/output-arm-image:/build/output-arm-image ghcr.io/solo-io/packer-plugin-arm-image:v0.2.5 build samples/raspiOS_cloud-init_armhf.json
```

results in

```bash
arm-image: output will be in this color.

==> arm-image: Image type: raspberrypi
==> arm-image: Retrieving Image
==> arm-image: Trying https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2021-05-28/2021-05-07-raspios-buster-armhf-lite.zip
==> arm-image: Trying https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2021-05-28/2021-05-07-raspios-buster-armhf-lite.zip?checksum=sha256%3Ac5dad159a2775c687e9281b1a0e586f7471690ae28f2f2282c90e7d59f64273c
==> arm-image: https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2021-05-28/2021-05-07-raspios-buster-armhf-lite.zip?checksum=sha256%3Ac5dad159a2775c687e9281b1a0e586f7471690ae28f2f2282c90e7d59f64273c => /root/.cache/packer/eb57a9ac054a8c6bae94d7bb16261accc9e9b78e.iso
==> arm-image: Copying source image.
==> arm-image: Image is a zip file.
==> arm-image: Unzipping 2021-05-07-raspios-buster-armhf-lite.img
    arm-image: Speed:   68.67 MB/s
    arm-image: Progress: 18.31%
    arm-image: Speed:   54.28 MB/s
    arm-image: Progress: 32.79%
    arm-image: Speed:   35.18 MB/s
    arm-image: Progress: 42.17%
    arm-image: Speed:   48.09 MB/s
    arm-image: Progress: 55.00%
    arm-image: Speed:   32.31 MB/s
    arm-image: Progress: 63.61%
    arm-image: Speed:   48.22 MB/s
    arm-image: Progress: 76.47%
    arm-image: Speed:   77.66 MB/s
    arm-image: Progress: 97.22%
==> arm-image: Growing partition to 4096 M (4294967296 bytes)
    arm-image: mapping output-arm-image/image
==> arm-image: losetup --show -f -P output-arm-image/image
==> arm-image: partitions: []
==> arm-image: no partitions defined
Build 'arm-image' errored after 2 minutes 8 seconds: step canceled or halted

==> Wait completed after 2 minutes 8 seconds

==> Some builds didn't complete successfully and had errors:
--> arm-image: step canceled or halted

==> Builds finished but no artifacts were created.
```

The setup is

```bash
$ docker --version
Docker version 20.10.11, build dea9396
$ cat /etc/issue
Ubuntu 20.04.3 LTS \n \l
```

Also for reference see [this old issue](https://github.com/solo-io/packer-plugin-arm-image/issues/105#issuecomment-950318311)

yuval-k, 2021-12-09:
Hi, I responded in the old issue, I think this will solve your problem:

> @groldo i think i see the issue, please add `-v /dev:/dev` to your docker run arguments (we updated the readme recently)

groldo, 2021-12-10:
Hi @yuval-k, I'm closing this as it now works like a charm 👍 .

Thanks for the great support and the awesome work.

closed 2021-12-10 by groldo

marcelomanzo, 2022-09-08:
I am facing the same issue, while using gitlab ci runner, even when using: **-v /dev:/dev**
build_os_dist:
  image: "docker:latest"
  stage: .pre
  services:
    - docker:dind
  script:
    - apk update
    - apk add git
    - git clone https://github.com/solo-io/packer-plugin-arm-image
    - cd packer-plugin-arm-image
    - docker run --rm --privileged -v **/dev:/dev** -v ${PWD}:/build:ro -v ${PWD}/packer_cache:/build/packer_cache -v ${PWD}/output-arm-image:/build/output-arm-image ghcr.io/solo-io/packer-plugin-arm-image build samples/raspbian_golang.json
  tags:
    - linux

Output...

==> arm-image: Copying source image.
==> arm-image: Image is a zip file.
==> arm-image: Unzipping 2020-02-13-raspbian-buster-lite.img
    arm-image: Speed:  148.38 MB/s
    arm-image: Progress: 40.25%
    arm-image: Speed:   91.92 MB/s
    arm-image: Progress: 65.03%
    arm-image: Speed:   55.48 MB/s
    arm-image: Progress: [80](https://gitlab.com/simracingstudio/srsapp/-/jobs/3001427196#L80).06%
==> arm-image: Growing partition to 4096 M (4294967296 bytes)
    arm-image: mapping output-arm-image/image
==> arm-image: losetup --show -f -P output-arm-image/image
==> arm-image: partitions: []
==> arm-image: no partitions defined
Build 'arm-image' errored after 33 seconds 467 milliseconds: step canceled or halted
==> Wait completed after 33 seconds 467 milliseconds
==> Some builds didn't complete successfully and had errors:
--> arm-image: step canceled or halted
==> Builds finished but no artifacts were created.
Cleaning up project directory and file based variables