solo-io/packer-plugin-arm-image #124: Error loading configuration:  fork/exec /build/packer-plugin-arm-image: exec format error

mikemorris, 2021-12-28:
I'm trying to get this plugin to run on macOS, and I'm hitting the following error on both the Vagrant and Docker build options documented in the README.

```
    default:  Error loading configuration:
    default:
    default: fork/exec /root/.packer.d/plugins/packer-plugin-arm-image: exec format error
```

```
docker run \
        --rm \
        --privileged \
        -v /dev:/dev \
        -v {$PWD}:/build:ro \
        -v {$PWD}/packer_cache:/build/packer_cache \
        -v {$PWD}/output-arm-image:/build/output-arm-image \
        -e PACKER_CACHE_DIR=/build/packer_cache \
        packer-builder-arm build samples/raspbian_golang.json
running /bin/packer
 Error loading configuration:

fork/exec /build/packer-plugin-arm-image: exec format error
```

The standalone build no longer works due to the switch from `kpartx` to `losetup` in #115. Is this perhaps an issue with a newer version of Packer config syntax, or a need to upgrade these plugins to use the [new plugin SDK for Packer v1.7 and later](https://www.packer.io/guides/1.7-plugin-upgrade)?

yuval-k, 2021-12-28:
do you have the new M1 mac?

yuval-k, 2021-12-28:
if your mac is indeed M1 the solution might be to publish multi-arch docker images

mikemorris, 2021-12-29:
Nope, I'm on an Intel Mac running macOS Monterey 12.1 with Packer 1.7.8

mikemorris, 2022-02-12:
This appears to actually just be an issue with where the binary built with `go build` is placed (not sure if it's a change in a newer version of Go and/or specific to macOS). https://github.com/solo-io/packer-plugin-arm-image/blob/52faad571c70656187238268e2032160b74fe695/Dockerfile#L38 assumes the binary will be placed into `./build` but on my machine it's instead placed into the repo root, not a subfolder. Moving the binary into the expected directory before building the Docker image avoids this issue.
```
mkdir build
mv packer-builder-arm ./build
```

closed 2022-02-12 by mikemorris

nfx, 2023-01-07:
@mikemorris for fun i'm trying to enable native macOS support in https://github.com/solo-io/packer-plugin-arm-image/pull/155