solo-io/packer-plugin-arm-image #130: Question: Why do we need to share the `/dev` folder of the host machine?

xserrat, 2022-03-12:
Hi! I have the question of the subject regarding the need for the `/dev` folder as a volume mapping when running the docker container. I saw that without this mapping, the process cannot find any partition. Regarding the `builder` configuration, I've defined the following:

```hcl
source "arm-image" "ubuntu" {
  iso_url =  "file:///workspace/input-images/ubuntu-20.04.4/ubuntu-20.04.4-preinstalled-server-arm64+raspi.img.zip"
  iso_checksum =  "d75844d7d6606681b6ecbb523c97672626edb40497ac11ec70eed520a2d0459a"
  image_mounts = ["/boot", "/"]
  output_filename = "/workspace/output-arm-images/ubuntu-20.04"
}
```

The image corresponds to the ubuntu server image for Raspberry. I've added these two mounting points `/boot` and `/`, but I'm not sure if it's ok. 

Thank you for your time and great plugin :)

yuval-k, 2022-03-12:
IIRC we need access to the main loop device to mount the partitions using `losetup`

xserrat, 2022-03-12:
Thanks for your fast reply @yuval-k !

closed 2022-03-12 by xserrat