solo-io/packer-plugin-arm-image #135: raspbian_golang.json buster InRelease' changed its 'Suite' value from 'stable' to 'oldstable'

5t33, 2022-04-17:
I get the following error:
```
==> arm-image: E: Repository 'http://raspbian.raspberrypi.org/raspbian buster InRelease' changed its 'Suite' value from 'stable' to 'oldstable'
==> arm-image: E: Repository 'http://archive.raspberrypi.org/debian buster InRelease' changed its 'Suite' value from 'testing' to 'oldstable'
```

when I run:
```
docker build . -t packer-plugin-arm-image-local:04-17-22
docker run \
  --rm \
  --privileged \
  -v /dev:/dev \
  -v ${PWD}:/build:ro \
  -v ${PWD}/packer_cache:/build/packer_cache \
  -v ${PWD}/output-arm-image:/build/output-arm-image \
  -e PACKER_CACHE_DIR=/build/packer_cache \
  packer-plugin-arm-image-local:04-17-22 build samples/raspbian_golang.json
```

5t33, 2022-04-20:
To be clear, that image is built from the repo's master branch. I see the same issue from the image pulled from latest.

tvieregge, 2022-07-07:
It looks like something isn't handling the warning message about buster being depreciated

If you change the iso_url to a bullseye image everything works fine

This one worked for me: https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2022-04-07/