solo-io/packer-plugin-arm-image #141: Can Please confirm if md5 checksum is validated properly?

mikzuit, 2022-05-14:
I been trying to build from  https://osmc.tv/download/ images but   constantly give me the output bellow. Even though I checked with  wget after donwload the file and check the md5 matched successfully , so is not the website md5.

command used:

```
docker run \
  --rm \
  --privileged \
  -v /dev:/dev \
  -v ${PWD}:/build:ro \
  -v ${PWD}/packer_cache:/build/packer_cache \
  -v ${PWD}/output-arm-image:/build/output-arm-image \
  ghcr.io/solo-io/packer-plugin-arm-image build samples/raspbian_golang.json

```
snippet of my builder:

```
"builders": [{
    "type": "arm-image",
    "image_type":"raspberrypi",
    "iso_url": "https://ftp.fau.de/osmc/osmc/download/installers/diskimages/OSMC_TGT_rbp2_20220314.img.gz",
    "iso_checksum": "md5:175b87851dbf1143df351cfb4134dbdd",
    "target_image_size": 4294967296
  }],  
```

output recceived:

```
==> arm-image: error downloading Image: [Checksums did not match for /build/packer_cache/1a719beb123cb2282c07f70f69d7bfe8b39c9487.iso.
==> arm-image: Expected: 28a8495c4da0ece10c8b12f0be6eb5f0
==> arm-image: Got: 175b87851dbf1143df351cfb4134dbdd
==> arm-image: *md5.digest]
Build 'arm-image' errored after 1 minute 43 seconds: error downloading Image: [Checksums did not match for /build/packer_cache/1a719beb123cb2282c07f70f69d7bfe8b39c9487.iso.
Expected: 28a8495c4da0ece10c8b12f0be6eb5f0
Got: 175b87851dbf1143df351cfb4134dbdd
*md5.digest]
```

expected output:

` successfuly build `

thanks

yuval-k, 2022-05-21:
the message says expected `28a8495c4da0ece10c8b12f0be6eb5f0`... where is that coming from?

mikzuit, 2022-05-22:
> 

I can only asume:

1.  From OSMC's website , returned a bad checksum , but i try with wget and it return the proper hash.
2.  From the plugin, (packer-plugin-arm) . I have only seen examples with the builder and checksum sha256 , will check tomorrow another img using a md5. 
3.  Is docker in any way corrupting the md5 ?

yuval-k, 2022-05-22:
the checksum `175b87851dbf1143df351cfb4134dbdd` looks correct in the OSMC website. 
The message says `expected 28a8495c4da0ece10c8b12f0be6eb5f0` this usually means someone told packer to expect this hash... maybe it is in one of the config files?

mikzuit, 2022-05-22:
> the checksum `175b87851dbf1143df351cfb4134dbdd` looks correct in the OSMC website. The message says `expected 28a8495c4da0ece10c8b12f0be6eb5f0` this usually means someone told packer to expect this hash... maybe it is in one of the config files?

If I run a search for that hash (28a8495c4da0ece10c8b12f0be6eb5f0) nothing found on the project codebase.

what is is line telling?
`==> arm-image: error downloading Image: [Checksums did not match for /tmp/getter2272480967/temp. `

I wish I knew a way to search inside that docker container but i can't due to that build limitations, I guess I will have to rebuild a new one with required bash at least.

yuval-k, 2022-07-10:
i'm not 100% sure, the part the downloads the images is re-used from upstream packet...