solo-io/packer-plugin-arm-image #145: ARM "native" logic creating invalid images

llamasoft, 2022-08-22:
Attempting to build an ARM image from an ARM64 device results in an unusable output image.  This appears to be a result of #30.
The relevant code is here: https://github.com/solo-io/packer-plugin-arm-image/blob/52faad571c70656187238268e2032160b74fe695/pkg/builder/builder.go#L304-L310

When building an ARM image on an ARM64 device, qemu is never initialized.  As a result, the image sees the host's `aarch64` machine type and installs packages for the wrong architecture.  
Even when building for the same architecture as the target image, not using qemu can result in breaking encapsulation.  For example, chroot alone isn't enough to prevent changing the host machine's hostname: https://unix.stackexchange.com/q/588789

llamasoft, 2022-08-25:
I did some poking around and unfortunately, I don't think we can determine from the provided image if it's supposed to be for ARM or ARM64.  Specifically, the RaspiOS boot partition includes _all_ kernels, not just one or the other.

I'm thinking the best approach might be adding an option for the target image architecture and have it default to ARM.  If the target arch matches GOARCH, then it can skip qemu.

I'll see if I can whip up a PR for this.  It might take me a little bit because I need to learn go first. 😁

yuval-k, 2022-08-27:
i like the approach, just keep in mind that we should try and make the PR backwards compatible - if the new target option is missing, fallback to current logic

yuval-k, 2022-10-18:
fixed by #146

closed 2022-10-18 by yuval-k