solo-io/packer-plugin-arm-image #147: no partitions defined while using dind (docker in a docker) with gitlab ci runner

marcelomanzo, 2022-09-08:
I am facing the same issue, while using gitlab ci runner, even when using: -v /dev:/dev
build_os_dist:
image: "docker:latest"
stage: .pre
services:
- docker:dind
script:
- apk update
- apk add git
- git clone https://github.com/solo-io/packer-plugin-arm-image
- cd packer-plugin-arm-image
- docker run --rm --privileged -v /dev:/dev -v {PWD}/packer_cache:/build/packer_cache -v ${PWD}/output-arm-image:/build/output-arm-image ghcr.io/solo-io/packer-plugin-arm-image build samples/raspbian_golang.json
tags:
- linux

Output...

==> arm-image: Copying source image.
==> arm-image: Image is a zip file.
==> arm-image: Unzipping 2020-02-13-raspbian-buster-lite.img
arm-image: Speed: 148.38 MB/s
arm-image: Progress: 40.25%
arm-image: Speed: 91.92 MB/s
arm-image: Progress: 65.03%
arm-image: Speed: 55.48 MB/s
arm-image: Progress: 80.06%
==> arm-image: Growing partition to 4096 M (4294967296 bytes)
arm-image: mapping output-arm-image/image
==> arm-image: losetup --show -f -P output-arm-image/image
==> arm-image: partitions: []
==> arm-image: no partitions defined
Build 'arm-image' errored after 33 seconds 467 milliseconds: step canceled or halted
==> Wait completed after 33 seconds 467 milliseconds
==> Some builds didn't complete successfully and had errors:
--> arm-image: step canceled or halted
==> Builds finished but no artifacts were created.
Cleaning up project directory and file based variables

shantanoo-desai, 2022-09-12:
I faced the same problem. Make sure you are not overwritten any previously created images that may exist in the `images` directory. For me I created an Ubuntu 20.04 image and the next time I was creating an Ubuntu 22.04 Image but I forgot to save the `.img` to a different file which was causing this error. I make use of packer-variables to make it more automated.

lucymhdavies, 2022-10-04:
Aha, this looks the same as my issue, so I don't think this is specific to DIND

In my case, running natively on an RPi (i.e. outside of Docker)

```
source "arm-image" "raspios" {
  iso_url      = "https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2022-04-07/2022-04-04-raspios-bullseye-arm64-lite.img.xz"
  iso_checksum = "sha256:35f1d2f4105e01f4ca888ab4ced6912411e82a2539c53c9e4e6b795f25275a1f"
  target_image_size = 3222000000
}
```

```
arm-image.raspios: output will be in this color.

==> arm-image.raspios: Image type: raspberrypi
==> arm-image.raspios: Retrieving Image
==> arm-image.raspios: Trying https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2022-04-07/2022-04-04-raspios-bullseye-arm64-lite.img.xz
==> arm-image.raspios: Trying https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2022-04-07/2022-04-04-raspios-bullseye-arm64-lite.img.xz?checksum=sha256%3A35f1d2f4105e01f4ca888ab4ced6912411e82a2539c53c9e4e6b795f25275a1f
==> arm-image.raspios: https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2022-04-07/2022-04-04-raspios-bullseye-arm64-lite.img.xz?checksum=sha256%3A35f1d2f4105e01f4ca888ab4ced6912411e82a2539c53c9e4e6b795f25275a1f => /root/.cache/packer/9d8ad4fe8b6f43457efbb1512fc630e91f8b50ba.iso
==> arm-image.raspios: Copying source image.
==> arm-image.raspios: Image is a xz file.
    arm-image.raspios: Speed:   87.07 MB/s
    arm-image.raspios: Speed:   31.72 MB/s
    arm-image.raspios: Speed:   23.69 MB/s
    arm-image.raspios: Speed:   22.56 MB/s
    arm-image.raspios: Speed:   20.01 MB/s
    arm-image.raspios: Speed:    8.31 MB/s
    arm-image.raspios: Speed:   13.33 MB/s
    arm-image.raspios: Speed:   22.38 MB/s
    arm-image.raspios: Speed:    8.96 MB/s
    arm-image.raspios: Speed:   16.15 MB/s
    arm-image.raspios: Speed:   13.78 MB/s
    arm-image.raspios: Speed:   19.24 MB/s
    arm-image.raspios: Speed:   77.70 MB/s
==> arm-image.raspios: Growing partition to 3072 M (3222000000 bytes)
    arm-image.raspios: mapping output-raspios/image
==> arm-image.raspios: losetup --show -f -P output-raspios/image
==> arm-image.raspios: partitions: []
==> arm-image.raspios: no partitions defined
Build 'arm-image.raspios' errored after 1 minute 41 seconds: step canceled or halted
```

@shantanoo-desai's suggestion
> Make sure you are not overwriting any previously created images that may exist in the images directory. For me I created an Ubuntu 20.04 image and the next time I was creating an Ubuntu 22.04 Image but I forgot to save the .img to a different file which was causing this error.

doesn't apply in my case, because I'm running this packer build from a completely fresh directory every time