solo-io/packer-plugin-arm-image #150: Packer unnecessarily injecting environment variables from host with shell provisioner

lucymhdavies, 2022-10-04:
When running scripts using the shell provisioner, environment variables from the Packer's parent process are available within running scripts.

From https://www.packer.io/docs/provisioners/shell
There are a bunch of configs related to how environment variables should be injected into a provisioner from Packer, and documented behaviour would appear to be that host env vars shouldn't end up inside the child process unless explicitly configured.

This is how I've seen it work with other plugins, but because of how this particular plugin works I can see how we've ended up with this happening.

This plugin uses v0.2.11 of the plugin SDK
https://github.com/solo-io/packer-plugin-arm-image/blob/master/go.mod#L7

From https://github.com/hashicorp/packer-plugin-sdk/blob/main/CHANGELOG.md, v0.2.12 added some more configuration around env vars, so it _may_ be as simple as updating to the latest version of the SDK and it'll be fixed automatically by that. Or it may be more complex, requiring plugin-specific logic. I'm not entirely sure.

Either way, here's an example packer config:

```
packer {
  required_plugins {
    arm-image = {
      # https://github.com/solo-io/packer-plugin-arm-image
      source  = "github.com/solo-io/arm-image"
      version = "<= 1.0.0"
    }
  }
}

source "arm-image" "raspios" {
  iso_url      = "https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2022-04-07/2022-04-04-raspios-bullseye-arm64-lite.img.xz"
  iso_checksum = "sha256:35f1d2f4105e01f4ca888ab4ced6912411e82a2539c53c9e4e6b795f25275a1f"
}

build {
  sources = ["source.arm-image.raspios"]

  provisioner "shell" {
    script = "provision.sh"
  }
}
```

`provision.sh`

```
#!/bin/bash

# Debugging
set -ex
env | sort

# ... and a bunch of things after that we don't care about for now
```

Packer output:

```
arm-image.raspios: output will be in this color.

==> arm-image.raspios: Image type: raspberrypi
==> arm-image.raspios: Retrieving Image
==> arm-image.raspios: Trying https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2022-04-07/2022-04-04-raspios-bullseye-arm64-lite.img.xz
==> arm-image.raspios: Trying https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2022-04-07/2022-04-04-raspios-bullseye-arm64-lite.img.xz?checksum=sha256%3A35f1d2f4105e01f4ca888ab4ced6912411e82a2539c53c9e4e6b795f25275a1f
==> arm-image.raspios: https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2022-04-07/2022-04-04-raspios-bullseye-arm64-lite.img.xz?checksum=sha256%3A35f1d2f4105e01f4ca888ab4ced6912411e82a2539c53c9e4e6b795f25275a1f => /root/.cache/packer/9d8ad4fe8b6f43457efbb1512fc630e91f8b50ba.iso
==> arm-image.raspios: Copying source image.
==> arm-image.raspios: Image is a xz file.
...
    arm-image.raspios: mapping output-raspios/image
==> arm-image.raspios: losetup --show -f -P output-raspios/image
==> arm-image.raspios: partitions: [/dev/loop1p1 /dev/loop1p2]
    arm-image.raspios: Mounting: /dev/loop1p2
    arm-image.raspios: Mounting: /dev/loop1p1
==> arm-image.raspios: Mounting additional paths within the chroot...
    arm-image.raspios: Mounting: /proc
    arm-image.raspios: Mounting: /sys
    arm-image.raspios: Mounting: /dev
    arm-image.raspios: Mounting: /dev/pts
    arm-image.raspios: Mounting: /proc/sys/fs/binfmt_misc
==> arm-image.raspios: Provisioning with shell script: provision.sh
==> arm-image.raspios: + env
==> arm-image.raspios: + sort
    arm-image.raspios: HOME=/root
    arm-image.raspios: INVOCATION_ID=077ab4453d2c4305bc165e9f6bf32724
    arm-image.raspios: JOURNAL_STREAM=8:20164
    arm-image.raspios: LANG=en_GB.UTF-8
    arm-image.raspios: LOGNAME=root
    arm-image.raspios: NOMAD_ALLOC_DIR=/opt/nomad/alloc/875e21c9-1832-e784-bbd3-ea8949b654d6/alloc
    arm-image.raspios: NOMAD_ALLOC_ID=875e21c9-1832-e784-bbd3-ea8949b654d6
    arm-image.raspios: NOMAD_ALLOC_INDEX=0
    arm-image.raspios: NOMAD_ALLOC_NAME=build/periodic-1664892598.packer[0]
    arm-image.raspios: NOMAD_CPU_LIMIT=2000
    arm-image.raspios: NOMAD_DC=davnet
```

And those `NOMAD_` env vars above come from the parent process.

I'm using:
```
+ ./local/packer version
Packer v1.8.3

+ ./local/packer plugins installed
/root/.config/packer/plugins/github.com/solo-io/arm-image/packer-plugin-arm-image_v0.2.6_x5.0_linux_arm64
```

Lemme know if there's any further information you need, and if there's much we can do about this.

yuval-k, 2022-10-06:
thanks for the detailed report, i'll try to take a look at this in the next few days

yuval-k, 2022-10-26:
just pushed a version with updated deps, hopefully this helps!

lucymhdavies, 2022-11-12:
Giving it a try :)

With the old version of the plugin...

```
+ ./local/packer plugins installed
/root/.config/packer/plugins/github.com/solo-io/arm-image/packer-plugin-arm-image_v0.2.6_x5.0_linux_arm64
```

here's an env var which should not be there:

```
+ ./local/packer build .

...

==> arm-image.raspios: + env
==> arm-image.raspios: + sort
    arm-image.raspios: NOMAD_DC=davnet
    arm-image.raspios: NOMAD_GROUP_NAME=packer
...
```

and with the latest... 

```
/root/.config/packer/plugins/github.com/solo-io/arm-image/packer-plugin-arm-image_v0.2.6_x5.0_linux_arm64
/root/.config/packer/plugins/github.com/solo-io/arm-image/packer-plugin-arm-image_v0.2.7_x5.0_linux_arm64
```

```
+ ./local/packer build .

...

Error: 1 error(s) occurred:

* embedded qemu is not available - currently, embedded qemu is only available for linux amd64. please download qemu-user-static manually

  on raspios.pkr.hcl line 15:
  (source code not available)

==> Wait completed after 9 microseconds
```

that's... odd.

lucymhdavies, 2022-11-12:
Lemme just...

```
# apt install qemu-user-static
```

and try again...

```
+ ./local/packer build .

...

    arm-image.raspios: NOMAD_DC=davnet
    arm-image.raspios: NOMAD_GROUP_NAME=packer
```

so the unexpected env vars are still there