solo-io/packer-plugin-arm-image #153: Multiple output files release

UnoSD, 2022-10-22:
I notice that with the last release, two files are in the archive:

```
packer-plugin-arm-image_v0.2.6_linux_amd64
packer-plugin-arm-image_v0.2.6_x5.0_linux_amd64
```

can you please clarify what the `x5.0` and without means?

Thank you

yuval-k, 2022-10-24:
I see the problem - 5 is the packer api version. this is the one to use; ignore the one without the 5.0 for now; i need to rename it to `flahser`

yuval-k, 2022-10-24:
(for reference, this is where the names are decided https://github.com/solo-io/packer-plugin-arm-image/blob/master/.goreleaser.yml)
note that you the new `required_plugin` block (see [example](https://github.com/solo-io/packer-plugin-arm-image/blob/4639fcce4d39ce7ecadf6bf2c7f6bd55642c15e8/samples/k8s/plugin.pkr.hcl#L2)) to have packer automatically download it for you

UnoSD, 2022-10-24:
thank you @yuval-k for the clarification and the suggestion on the `required_plugin`, really useful.

in addition, packer automatically selected the correct binary when executed.

closed 2022-10-24 by UnoSD