solo-io/packer-plugin-arm-image #167: BtrFS support

thomastthai, 2023-12-06:
How well is `btrfs` supported for the Raspberry Pi if the root partition was converted to `btrfa` from `ext4`?

`btrfs` partition resize is unlikely to work with `resizefs` and would require the `btrfs-progs` package.

jonathanmarchuot, 2023-12-06:
This is also currently a headache of mine -- patching Ubuntu images that Libre Computer is publishing with btrfs supported.

Currently, the way I've found around this, is by making sure your json gets rid of the "target_image_size", and then resize your source image prior to running packer.

```
truncate --size=+1400M ./original.img

sudo losetup --show -f -P ./original.img
#	'/dev/loop3'

sudo parted /dev/loop3
#	'print'
#		Disk /dev/loop3: 7213MB
#		part 2 end 2809MB size 2540MB
#
#      ((take the disk size '7213' above, then resize the partition to take the full space))
#	'resizepart 2'
#		End?  [2809MB]?
#		'7213MB'
#
#	'print'
#		Disk /dev/loop3: 7213MB
#		part 2 end 7213MB size 6944MB
#
#	'quit'

sudo mkdir -p /btrfs
sudo mount /dev/loop3p2 /btrfs

sudo btrfs filesystem resize max /btrfs
sudo btrfs filesystem show
sudo btrfs check --force /dev/loop3p2

sudo umount /btrfs
sudo losetup -d /dev/loop3

shasum -a 256 ./original.img

# proceed with packer
```

Good luck. Would love to see btrfs resizing supported in the future!

jonathanmarchuot, 2023-12-06:
This is also currently a headache of mine -- patching Ubuntu images that Libre Computer is publishing with btrfs supported. I would love to see btrfs resizing supported by the project in the future!

Currently, the way I've found around this, is by making sure your json gets rid of the "target_image_size", and then resize your source image prior to running packer.

```
truncate --size=+1400M ./original.img

sudo losetup --show -f -P ./original.img
#	'/dev/loop3'

sudo parted /dev/loop3
#	'print'
#		Disk /dev/loop3: 7213MB
#		part 2 end 2809MB size 2540MB
#
#      ((take the disk size '7213' above, then resize the partition to take the full space))
#	'resizepart 2'
#		End?  [2809MB]?
#		'7213MB'
#
#	'print'
#		Disk /dev/loop3: 7213MB
#		part 2 end 7213MB size 6944MB
#
#	'quit'

sudo mkdir -p /btrfs
sudo mount /dev/loop3p2 /btrfs

sudo btrfs filesystem resize max /btrfs
sudo btrfs filesystem show
sudo btrfs check --force /dev/loop3p2

sudo umount /btrfs
sudo losetup -d /dev/loop3

shasum -a 256 ./original.img
```

And then take your "new" original.img with your new shasum and run packer against that. Good luck!

thomastthai, 2023-12-06:
@jonathanmarchuot, thanks for sharing your workaround. Are you doing the workaround manually or automated via code?